evereux/pycatia #99: Incorrect VBA Function Output[BUG]

Mebiza, 2021-09-08:
**Describe the bug**
Some functions would return an error on the output due to a missing comma separating output parameters

**To Reproduce**
Call get_point_at_param on a Curve2D object.
start_point = curve.get_point_at_param(sp) -> Error

**Expected behavior**
Te function should return the Point object for the given parameter value on the curve

**Screenshots**
pycatia code for get_point_at_param:
[image]
How it should be:
[image]

Adding the comma before oPoint solves the issue

**Desktop (please complete the following information):**
 - OS: Windows 10
 - CATIA V5 version V5-6R2018
 - pycatia version [e.g. pycatia 0.4.4] i believe it still exists on 0.5.1

**Additional context**
Add any other context about the problem here.

evereux, 2021-09-27:
Thanks for this. I'll get this done in the next few days.

evereux, 2021-09-29:
Released new version 0.5.2 that fixes this.

closed 2021-09-29 by evereux