evereux/pycatia #100: Edge Search selection for selected part in product level not working

tcrakshi, 2021-09-17:
i am working in product level, in each product i need to select specific part and  in that part i need to select all the edges.

its selecting when i am given  - selectionSet.search("Type=Edge,all")
its not selecting when i am given  - selectionSet.search("Type=Edge,all")

may code same as below:
``` Python Code
from pycatia import catia
import sys, time, os
from pycatia.enumeration.enumeration_types import cat_work_mode_type
from pycatia.product_structure_interfaces import product

caa = catia()
document = caa.active_document
product1 = document.product
Prod = product1.products
spescif_prod = Prod.item(1)

#for item in product1:
caa.active_document.selection.clear()
caa.active_document.selection.add(spescif_prod)

time.sleep(2)
selectionSet = caa.active_document.selection
selectionSet.search("Type=Edge,sel")
```

after this code run i am ending up with no selection in object(No error )

evereux, 2021-09-29:
Does it work in just a part? 

This seems remarkably similar to [#101](https://github.com/evereux/pycatia/issues/101).

evereux, 2022-06-03:
Closing due to lack of feedback.

closed 2022-06-03 by evereux