evereux/pycatia #101: select only specific type edge under edge selection

tcrakshi, 2021-09-17:
My requirement is to select only circle type of edge on part. 
Part is under assembly, means to say my macro need to work  in Product workbench.

i found below type of edges 
``
RectilinearMonoDimFeatEdge
MonoDimFeatEdge
RectilinearBiDimFeatEdge
TriDimFeatEdge
BiDimFeatEdge
RectilinearTriDimFeatEdge
``

evereux, 2021-09-29:
How would you do this using CATIA itself without using the selection filter toolbar?

The only way I can think of is to use the search feature. To do this in an automated way you would use the [selection.search](https://github.com/evereux/pycatia/blob/master/examples/example_019.py#L27) feature as shown in this [example](https://github.com/evereux/pycatia/blob/master/examples/example_019.py#L27).

evereux, 2022-06-03:
Closing due to lack of feedback.

closed 2022-06-03 by evereux