evereux/pycatia #103: Cylindrical Face Origin and direction

tcrakshi, 2021-09-20:
i an trying to fetch the Get_origin and  and Get_direction method on Cylindrical face, but i am geeting addres of location.

below is my code 
``` python Code
from pycatia import catia
import sys, time, os
from pycatia.enumeration.enumeration_types import cat_work_mode_type
from pycatia.product_structure_interfaces import product
from pycatia.mec_mod_interfaces import face
from pycatia.mec_mod_interfaces import cylindrical_face

def Print_error():
    exc_type, exc_obj, exc_tb = sys.exc_info()
    fname = os.path.split(exc_tb.tb_frame.f_code.co_filename)[1]
    print(exc_type, fname,  "Line: "+ str(exc_tb.tb_lineno))

try:
    caa = catia()
    document = caa.active_document
    product1 = document.part    
    selectionSet = caa.active_document.selection
    selectionSet.search("Topology.Face,sel")
    
    surface_type = []
    surface_type_index = []
    print (selectionSet.count)
    
    for i in range(1, selectionSet.count):
        if selectionSet.item(i).type==  "CylindricalFace":
            Sel_object = selectionSet.item(i).value
            sel_Com_object = cylindrical_face.CylindricalFace(Sel_object.com_object)
            surface_type.append(sel_Com_object)

    selectionSet.clear()
    for item in surface_type:
        C_Face = item.cylindrical_face
        list_O = tuple()
        list_O = item.get_origin
        print(list_O)
        print()
        
except Exception as e:
    print(e)
    Print_error()
    caa.message_box("No Document is Opened to run this appliction")     

output i am geeting same as leow tuple
```
<bound method CylindricalFace.get_origin of CylindricalFace(name="Selection_RSur:(Face:(Brp:(ThickSurface.7;2:(Brp:(GSMFillet.306;(Brp:(GSMExtrude.135;0:(Brp:(Sketch.103;1)));Brp:(GSMOffset.193;(Brp:(GSMBiDim.193;1)))))));None:();Cf11:());Remove.4_ResultOUT;Z0;G7592)")>

```

my expectation is to fetch the coordinate value.
could you please help 

```

evereux, 2021-12-02:
`CylindricalFace.get_origin` is a method, not a property, so should be called up as `item.get_origin()`. [CylindricalFace](https://github.com/evereux/pycatia/blob/master/pycatia/mec_mod_interfaces/cylindrical_face.py).

However, the method required mapping to a VBA to work properly. I've fixed that now in version `0.5.3`.

evereux, 2021-12-02:
Also, please describe your CATParts / CATProducts, or provide screenshots, so I can reproduce them. It took me a while to figure out how to generate a cylindrical face. Figured a padded circle -> cylinder would do, but no ...

evereux, 2022-06-03:
Closing due to lack of feedback.

closed 2022-06-03 by evereux