evereux/pycatia #104: how to fetch the information of Gemetrical set , bodies and axis systme of a part - if part is under product

tcrakshi, 2021-09-24:
hi,

i have root product, i want loop and fetcch part and need to get the how many gemetrical set are there and i want create  one point.

Note: The part is located under product

do we have any example?

Tian-Jionglu, 2021-09-25:
Hi,
1. get `Collection` object `product.products`, then you can loop;
2. get `Collection` object `part.hybrid_bodies`, use `len()` function to get the number of geometrical sets;

tcrakshi, 2021-09-27:
if possible could you please provide some snippet or examples
```
from pycatia import catia
import sys, time, os

caa = catia()
document = caa.active_document
product1 = document.product
#
product_col = product1.products

for item in product_col:
    print(dir(item))
    ### Struck here to get the Part level hybridshape factory from Item

```

Tian-Jionglu, 2021-09-29:
try this:
```
from pycatia import catia
from pycatia.mec_mod_interfaces.part_document import PartDocument
import sys, time, os

caa = catia()
document = caa.active_document
product1 = document.product

product_col = product1.products

for prd in product_col:
    if prd.type == "CATPart":

        # since part and product and siblings, it must be get Document first
        part_doc = PartDocument(prd.reference_product.parent.com_object) # this clause is not pretty, since prd.reference_product.parent can only return AnyObject at present
        part = part_doc.part
        hybrid_bodies = part.hybrid_bodies
        print(len(hybrid_bodies))
```

tcrakshi, 2021-09-29:
really helped, thanks

closed 2021-09-29 by tcrakshi