evereux/pycatia #105: [Example REQUEST]  Can I get an example of using the GSD split operation?

DIEOM, 2021-11-15:
**Is your feature request related to the CAA V5 Automation API?**
If yes, please describe where in the API documentation that feature is described.

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

I learning about CCA V5 Automation by pycatia

But I don't know how to use a function that is not in the example.

So, Can I get an example of using the GSD split operation? 

Thanks

evereux, 2021-12-02:
Unfortunately I can't provide examples for this as I've never used it.

Can you provide some examples of how this would be done in CATScript / VBA etc? If you do some googling maybe you'll find some. That in conjunction with what you'll learn running through the examples you should get a good idea of what needs to be done using pycatia.

If you can't figure it out post what you have and I'll see what I can do to help. But, I need something to work with. I simply don't have the time / resources to figure out and provide examples for everything pycatia is capable of.

evereux, 2022-04-19:
I've now created a small example script to show the usage of this feature. See [example_030](https://github.com/evereux/pycatia/blob/master/examples/example_030.py)

closed 2022-04-19 by evereux