evereux/pycatia #106: Extract Surface

tcrakshi, 2021-12-24:
i am working on product, based on user selection (face/Surface), I need to extract that surface to other part.

could anyone help with some example

tcrakshi, 2022-01-06:
``` python code

from pycatia import catia
from pycatia.enumeration.enumeration_types import cat_work_mode_type
from pycatia.product_structure_interfaces import product
from pycatia.mec_mod_interfaces import edge
from pycatia.mec_mod_interfaces import part
from pycatia.hybrid_shape_interfaces import point
from pycatia.hybrid_shape_interfaces import line
from pycatia.hybrid_shape_interfaces import plane
from pycatia.mec_mod_interfaces import face
from pycatia.mec_mod_interfaces.part_document import PartDocument

def get_part_From_Product(iProdcut):
    try:
        part_doc = PartDocument(iProdcut.reference_product.parent.com_object) # this clause is not pretty, since prd.reference_product.parent can only return AnyObject at present
        opart = part_doc.part
        return opart
    except Exception as e:
        print(e)

def get_Type_Document():
    document = caa.active_document
    try:
        product1 = document.part
        print ("Working Document is part")
        print(product1)
        return product1, "Part"
    except:
        try:
            product1 = document.product
            product1.count_children
            Product_ID = product1.part_number
            caa.active_document.selection.clear()
            product1.apply_work_mode(cat_work_mode_type.index("DESIGN_MODE"))
            print ("Working Document is product")
            return product1 , "Product" , Product_ID
        except Exception as e:
            print(e)
            
def Create_HSF(iProduct):
    try:
        GAEpart = get_part_From_Product(iProduct)
        hybrid_Sahpe_factory = GAEpart.hybrid_shape_factory
        hybrid_Bodies = GAEpart.hybrid_bodies
        return GAEpart, hybrid_Sahpe_factory, hybrid_Bodies 
    except Exception as e:
        print(e)

def Create_GeometricalSet_inPart(iLocation , iString ):
    hybrid_bodies = iLocation.hybrid_bodies
    oGemetricalSet = hybrid_bodies.add()
    oGemetricalSet.name = iString
    return oGemetricalSet
           
def New_GAE_Part_cb(): ### it will create new part inside tthe root product 
    global GAE_Product
    global hsf
    try:
        if document_type == "Product":
            caa.active_document.selection.clear()
            GAE_Product = root_product.products.add_new_component("Part", "")
            try:
#                GAE_Product.name = GAE_part_name
                global GAE_hsf
                global GAE_hybridbody
                global GAE_part
                global GAE_point_geo_Set
                GAE_part, GAE_hsf, GAE_hybridbody = Create_HSF(GAE_Product)
                GAE_point_geo_Set = Create_GeometricalSet_inPart(GAE_part, "Surface")
                GAE_part.update()
            except Exception as e:
                print(e)
                caa.message_box("GAE Point part alreday exist in this assembly")
    except Exception as e:
        print(e)

def Select_surface_cb():
    try:
        if GAE_Product is not  None:
            caa.active_document.selection.clear()
            object_filter = ("PlanarFace","Plane","Face",)
            selectionSet = caa.active_document.selection
            status = selectionSet.select_element2(object_filter,"Select Surface,  you want create the point" , False )
            if status == "Normal":
                Sel_object = selectionSet.item(1).value
                Face_object = face.Face(Sel_object.com_object)
                print(Face_object)
                Extract_surface3 = GAE_hsf.add_new_datums(Face_object)
                GAE_point_geo_Set.append_hybrid_shape(Extract_surface3)
        else:
            caa.message_box("Please Select/Create GAE Point part before proceding")
    except Exception as e:
        print(e)

document_Name = None
try:
    caa = catia()
    root_product = None
    document_type = None
    document_Name = None
    root_product, document_type, document_Name = get_Type_Document()
#    MainDlg()
    if document_type == "Product" or document_type == "Part":
        document = caa.active_document
        spa_workbench = document.spa_workbench()
        New_GAE_Part_cb()
        Select_surface_cb()
except Exception as e:
    print(e)
```

evereux, 2022-03-13:
There seems to be a lot of irrelevant code here again and as a result it's time consuming trying to get to the root of your problem. I do want to help but it's very frustrating trying to deal with your help requests.

Now, do you know that what you're trying to do is possible purely using CAA automation?  If so, let's simplify this... 

Assume we have two parts already open. 
Part1 has a surface - we know it's name and can select it that way. 
Part2 is the target.
Now write me some code showing how you think this should work.

evereux, 2022-06-03:
I'm closing this due to lack of feedback.

closed 2022-06-03 by evereux