evereux/pycatia #107: Unresolved attribute reference 'product' for class 'Document'

Erstreben, 2021-12-29:
Hello,  when I write the code ,  I always have the problem of " Unresolved attribute reference 'product' for class 'Document'" or Unresolved attribute reference 'part' for class 'Document',
[image]
For this reason, Pycharm does not remind me of the method of "Part". So is it a BUG or I make it wrong?
Thank you!

evereux, 2022-01-02:
> For this reason, Pycharm does not remind me of the method of "Part". 

I'm not sure what you mean here?

> So is it a BUG or I make it wrong?

I'm not so sure I'd class it a BUG (at least not an issue that breaks anything) but more an area where I can improve type hinting? I have an idea of why pycharm is reporting this (I also use pycharm but ignore this warning, it never bothered me) and I'll take a look to see if this is something that can be improved at some point.

deloarts, 2022-12-03:
@Erstreben: The reason for this that `caa.active_document` returns the `Document` object from InfInterfaces framework, as described in the help file:

[image]

When you further inspect the `Document` object in the help file, you'll see there is no `Part` or `Product` property:

[image]

This is because the `Document` object is responsible for managing the file on disk (or database), not for managing e.g. product structures.

In VB you would set the product variable like this:
```vbs
Sub CATMain()
  Set ActiveDoc = CATIA.ActiveDocument
  Dim RootProduct As Product
  Set RootProduct = ActiveDoc.Product
  ...
End Sub
```

So in python you have to:
```python
from pycatia import catia
from pycatia.product_structure_interfaces.product_document import ProductDocument

caa = catia()
product_doc = ProductDocument(caa.active_document.com_object)
root_product = product_doc.product

print(root_product.full_name)
```

Imagine it like this: The active document could be anything (there are 8 types of documents), so you have to explicitly tell what document to expect. Like you have ProductDocument open and would use a property only available in a PartDocument, this wouldn't turn out that well.

ptm-tm, 2023-03-03:
i think mb catia will be a class with corresponding properties like 

> 
    def set_active_file(self):
        """
        Depending on which document exists, the corresponding document class is set.
        """
        if self.active_doc.is_part:
            self.active_file = self.active_doc.part()
        elif self.active_doc.is_product:
            self.active_file = self.active_doc.product()
            self.set_all_children()
        elif self.active_doc.is_drawing:
            self.active_file = self.active_doc.drawing_root()
>

or

ptm-tm, 2023-03-04:
can anyone share com3dx.pyc from 3dexpirience? path  Bxx\win_b64\code\pythonX.x\lib

evereux, 2023-03-04:
> can anyone share com3dx.pyc from 3dexpirience? path Bxx\win_b64\code\pythonX.x\lib

@ptm-tm - please raise a new issue for this request as it's not related to this issue.

evereux, 2023-09-14:
Closing this now as the documentation / examples have been updated to show how to initialise products / parts etc so you can get code completion in IDE's.

That said I'd still like to improve on how pycatia does this at some point in the very near future.