evereux/pycatia #108: [HELP] Set surface for create a line normal to surface.

miguelcaralb, 2022-03-04:
Hello,

I am trying to draw normal lines to the surface. I can't define the surface to pass it as a parameter to the module that creates this type of line. The code in VBA is as follows:

Set partDocument1 = CATIA.ActiveDocument
Set part1 = partDocument1.Part
Set hybridShapeFactory1 = part1.HybridShapeFactory
Set hybridBodies1 = part1.HybridBodies
**Set hybridBody1 = hybridBodies1.Item("SURFACE")
Set hybridShapes1 = hybridBody1.HybridShapes
Set hybridShapeExtract1 = hybridShapes1.Item("Master")
Set reference1 = part1.CreateReferenceFromObject(hybridShapeExtract1)**
Set parameters1 = part1.Parameters
Set hybridShapePointExplicit1 = parameters1.Item("Point.2 ( ELEMENT - *MASTER - *SET1 - *PT2 )")
Set reference2 = part1.CreateReferenceFromObject(hybridShapePointExplicit1)
Set hybridShapeLineNormal1 = hybridShapeFactory1.AddNewLineNormal(reference1, reference2, -20.000000, 20.000000, False)
hybridBody1.AppendHybridShape hybridShapeLineNormal1
part1.InWorkObject = hybridShapeLineNormal1
part1.Update 

Thank you in advance.

miguelcaralb, 2022-03-05:
My code in Python:

from pycatia import catia

caa = catia()
documents = caa.documents

document = caa.active_document
part = document.part

hybrid_bodies = part.hybrid_bodies

product = document.product
docSelection = document.selection

parametro = product.parameters
propiedad = product.user_ref_properties

"""
****************
surface_1 = SURFACE???
******************
"""

hybrid_body_points = hybrid_bodies.add()
hybrid_shape_factory = part.hybrid_shape_factory
point_1 = hybrid_shape_factory.add_new_point_coord(24, -298, -370)
hybrid_body_points.append_hybrid_shape(point_1)

line_1 = hybrid_shape_factory.add_new_line_normal(surface_1, point_1, 20, -20, True)
hybrid_body_points.append_hybrid_shape(line_1)

evereux, 2022-03-13:
Can you format your code per [Raising Issues / Asking Questions](https://github.com/evereux/pycatia/issues/28) please?

I think the step your missing, which is in your VBA Script, is creating reference objects from the items you've created.

This following example works for me.

My tree
```
Part
    PartBody
    ConstructionGeometry
        Surface.Master
        Point.OnSurface.1
````

```python

from pycatia import catia

caa = catia()
document = caa.active_document
part = document.part
hsf = part.hybrid_shape_factory

# get the Geometrical Sets
hybrid_bodies = part.hybrid_bodies
# get the Geometrical Set named "ConstructionGeometry"
construction_geometry = hybrid_bodies.get_item_by_name('ConstructionGeometry')
construction_shapes = construction_geometry.hybrid_shapes

# get the surface named "Surface.Master" in the Geometrical Set "ConstructionGeometry"
master_surface = construction_shapes.get_item_by_name('Surface.Master')
master_surface_reference = part.create_reference_from_object(master_surface)

# get the point named "Point.OnSurface.1" in the Geometrical Set "ConstructionGeometry"
point_on_surface = construction_shapes.get_item_by_name('Point.OnSurface.1')
point_on_surface_reference = part.create_reference_from_object(point_on_surface)

line_normal = hsf.add_new_line_normal(master_surface_reference, point_on_surface_reference, -100, 100, True)
construction_geometry.append_hybrid_shape(line_normal)

part.update()```

miguelcaralb, 2022-03-15:
That's fine! Thank you so much!

closed 2022-03-15 by miguelcaralb