evereux/pycatia #109: AttributeError: AddNewPlaneOffsetPt.com_object[BUG]

hoangminhq5310, 2022-03-09:
I have AttributeError: AddNewPlaneOffsetPt.com_object when creating plane offset point

`  File "D:\PythonProject\AKACoga\Framework\newakacoga\tests\steps\step1.py", line 18, in action
    hs_plane = Plane.add_plane_offset_pt(part, hsf, hb_test, xy_plane, hs_pt1)
  File "D:\PythonProject\AKACoga\Framework\newakacoga\akacoga\comm\plane.py", line 30, in add_plane_offset_pt
    plane = hsf.add_new_plane_offset_pt(plane, point)
  File "D:\PythonProject\AKACoga\Framework\newakacoga\env\lib\site-packages\pycatia\hybrid_shape_interfaces\hybrid_shape_factory.py", line 3864, in add_new_plane_offset_pt
    return HybridShapePlaneOffsetPt(
  File "D:\PythonProject\AKACoga\Framework\newakacoga\env\lib\site-packages\pycatia\hybrid_shape_interfaces\hybrid_shape_plane_offset_pt.py", line 39, in __init__
    super().__init__(com_object)
  File "D:\PythonProject\AKACoga\Framework\newakacoga\env\lib\site-packages\pycatia\hybrid_shape_interfaces\plane.py", line 45, in __init__
    self.plane = com_object
  File "D:\PythonProject\AKACoga\Framework\newakacoga\env\lib\site-packages\pycatia\hybrid_shape_interfaces\hybrid_shape_plane_offset_pt.py", line 78, in plane
    self.hybrid_shape_plane_offset_pt.Plane = reference_plane.com_object
  File "D:\PythonProject\AKACoga\Framework\newakacoga\env\lib\site-packages\win32com\client\dynamic.py", line 639, in __getattr__
    raise AttributeError("%s.%s" % (self._username_, attr))
AttributeError: AddNewPlaneOffsetPt.com_object`

I fix the error by going to hybridshape_plane_offset_pt.py and change from line 42 to this:
`    @property
    def ref_plane(self) -> Reference:
        """
        .. note::
            :class: toggle

            CAA V5 Visual Basic Help (2020-07-06 14:02:20.222384)
                | o Property Plane() As Reference
                | 
                |     Role: Get the reference plane.
                | 
                |     Parameters:
                | 
                |         oPlane
                |             reference Plane. 
                | 
                |     See also:
                |         Reference 
                |     Returns:
                |         HRESULT S_OK if Ok E_FAIL else return error code for C++
                |         Implementations 
                |     See also:
                |         HybridShapeFactory

        :return: Reference
        :rtype: Reference
        """

        return Reference(self.hybrid_shape_plane_offset_pt.Plane)

    @ref_plane.setter
    def ref_plane(self, reference_plane: Reference):
        """
        :param Reference reference_plane:
        """

        self.hybrid_shape_plane_offset_pt.Plane = reference_plane.com_object`

then it works perfectly OK
Can you guys have a quick check to see if you encounter similar error, and to see if my fix works?
Thank you!

evereux, 2022-03-13:
Can you fix the formatting of your post and provide a small sample script for me to run so i can test this please?

See [Raising Issues](https://github.com/evereux/pycatia/issues/28).

This greatly helps me understand the issue better.

hoangminhq5310, 2022-03-13:
you will get an error just by running this simple piece of code

`from pycatia import catia

if __name__ == '__main__':
    caa = catia()
    caa.documents.add('part')
    doc = caa.active_document
    part = doc.part
    hsf = part.hybrid_shape_factory
    spa = doc.spa_workbench()
    geo = part.hybrid_bodies.add()
    pt1 = hsf.add_new_point_coord(1, 2, 3)
    geo.append_hybrid_shape(pt1)
    part.update_object(pt1)
    plane = part.origin_elements.plane_xy
    new_plane = hsf.add_new_plane_offset_pt(plane, pt1)
    geo.append_hybrid_shape(new_plane)
    part.update_object(new_plane)`

evereux, 2022-03-13:
Thanks. That saves me time.

Please also fix your formatting.
Whilst doing that can you double check what files / line numbers you're referring to as they don't make sense to me. `hybridshape_plane_offset_pt.py` does not contain those lines you've quoted.

evereux, 2022-03-13:
I can confirm the bug.

I'll look at how the bug should be fixed in the coming days. Thanks for reporting! 👍