evereux/pycatia #110: [Solution] Exporting BOM with AssemblyConvertor

deloarts, 2022-03-25:
### Problem
I tried exporting the bill of material as Excel using the **get_item** method:

```
bom = product.get_item("BillOfMaterial")
bom.set_current_format(format)
bom.print("XLS", path, product)
```

This is failing with `AttributeError: 'AnyObject' object has no attribute 'SetCurrentFormat'` since the **get_item** method doesn't return `AssemblyConvertor`.

### Solution
I made changes to the _any_object.py_ file by adding the following import

`from pycatia.system_interfaces.cat_base_dispatch import CATBaseDispatch`

and by adding the following method to the `AnyObject` class:

```
def get_item_dispatch(self, id_name: str) -> CATBaseDispatch:
    return self.com_object.GetItem(id_name)
```

I was following the typehints of the _V5automation.chm_:

```
AnyObject (method)
...
Methods
  GetItem(CATBSTR IDName) As CATBaseDispatch
```

This way i was able to get the `AssemblyConvertor` object and thus the bill of material as Excel:

```
from pycatia.product_structure_interfaces.product import Product
from pycatia.product_structure_interfaces.assembly_convertor import AssemblyConvertor

def export_bom(product: Product, path: str, format: tuple):
    bom = AssemblyConvertor(product.get_item_dispatch("BillOfMaterial"))
    bom.set_current_format(format)
    bom.set_secondary_format(format)
    bom.print("XLS", path, product)
```

with the `format` being a tuple containing the keywords of the products properties ("PartNumber", "Revision", ...) or the names of user_ref_properties and the `path` is using windows style seperators `\\`.

### Note
Sometimes the export failes with an error (`-2147467259`). I haven't figured out what this error is, because I only encountered it once. After restarting my computer I wasn't able to reproduce the error again. But I don't think this error is related to the pywin32 dispatch, because I also got this error by running the equivalent catvbs:

```
Sub CATMain()
    CATIA.StatusBar = "Running BOM Export"
    Dim Path: Path = "C:\Users\i030\Desktop\bom.xls"
    Dim Product
    Dim BOM
    Dim BOMFormat(4)
    
    BOMFormat(0) = "Typ"
    BOMFormat(1) = "Menge"
    BOMFormat(2) = "Teilenummer"
    BOMFormat(3) = "Definition"

    Set Product = CATIA.ActiveDocument.Product
    Set BOM = Product.GetItem("BillOfMaterial")

    BOM.SetCurrentFormat BOMFormat
    BOM.SetSecondaryFormat BOMFormat
    BOM.Print "XLS", Path, Product
End Sub
```

### Discussion
I havent't tested it yet, but it should be possible, that this way (returning the `get_item` method as `CATBaseDispatch`) may work for all other objects that you try to get with the `get_item` method. Do you think this could be added or may this break something else?

evereux, 2022-04-12:
Thanks for the question @deloarts!

I've submitted a patch to the development branch for which you can see the link above. This change only affects the assembly_convertor.print method so there's no worry of it affecting anything else.

My quick and dirty code that worked:

```python

from pycatia import catia
from pycatia.product_structure_interfaces.product import Product
from pycatia.product_structure_interfaces.assembly_convertor import AssemblyConvertor
caa = catia()
document = caa.active_document
product = document.product
product = Product(product.com_object)

bom = product.get_item("BillOfMaterial")
assembly_convertor = AssemblyConvertor(bom.com_object)

details_top = ("Quantity", "Part Number", "Type", "Nomenclature", "Revision")
details_recap = ("Quantity", "Part Number")
assembly_convertor.set_current_format(details_top)
assembly_convertor.set_secondary_format(details_recap)

assembly_convertor.print("XLS", "C:\\Users\\evereux\\Documents\\lalala.xls", product)
```

Thanks for reporting this and going to the effort you have. It's appreciated. 

I think this is something worth adding to the examples too.

evereux, 2022-04-12:
> Sometimes the export failes with an error (`-2147467259`). I haven't figured out what this error is, because I only encountered it once. After restarting my computer I wasn't able to reproduce the [...]

Just been playing a bit with this and I can reproduce this.

* run script to generate xls.
* press `No` when asked to overwrite file (this will throw a similar error).
* run script again (error is thrown, file is still created).

I will continue to get the error until I restart CATIA. This isn't a bug in pycatia as far as I know. Whether the bug is in pywin32 or CATIA itself I don't know.