evereux/pycatia #114: Excuse me, how can I set the origin coordinate and coordinate axis of the sketch, instead of a simple PlaneXY of origin. elements? Thanks

Fox-in-the-Scaborough-Fair, 2022-08-01:
Excuse me, how can I set the origin coordinate and coordinate axis of the sketch, instead of a simple PlaneXY of origin elements? Thanks

evereux, 2022-08-18:
This isn't something I've ever done. Do you have an example using CATScript / VBA?

evereux, 2022-08-18:
I've tried recording a macro that creates a positioned sketch using lines and point as reference and looking at the source code but that step isn't recorded.

evereux, 2022-08-18:
It would appear you can't create new sketches that aren't sliding (defined with a single plane) with VBA (and thus pycatia): https://www.eng-tips.com/viewthread.cfm?qid=385399.

closed 2022-08-18 by evereux

Fox-in-the-Scaborough-Fair, 2022-08-18:
Thank you for your reply! I have solved the problem. I create a point and a normal, which decide a plane as my sketch reference. Then I can draw my sketch on this plane. Your code gives me a lot of help, thank you!

Fox-in-the-Scaborough-Fair, 2022-08-18:
`    hybridShapeFactory = part.HybridShapeFactory
    hybridShapePointCoord1 = hybridShapeFactory.AddNewPointCoord(sketch_plane.origin[0], sketch_plane.origin[1], sketch_plane.origin[2])
    hybridShapePointCoord2 = hybridShapeFactory.AddNewPointCoord(sketch_plane.origin[0] + sketch_plane.normal[0], sketch_plane.origin[1] + sketch_plane.normal[1], sketch_plane.origin[2] + sketch_plane.normal[2])
    reference1 = part.CreateReferenceFromObject(hybridShapePointCoord1)
    reference2 = part.CreateReferenceFromObject(hybridShapePointCoord2)
    hybridShapeLinePtPt = hybridShapeFactory.AddNewLinePtPt(reference1, reference2)
    reference3 = part.CreateReferenceFromObject(hybridShapeLinePtPt)
    reference4 = part.CreateReferenceFromObject(hybridShapePointCoord1)
    hybridShapePlaneNormal = hybridShapeFactory.AddNewPlaneNormal(reference3, reference4)
    hybridShapePlaneNormal.name = plane_name

    body.InsertHybridShape(hybridShapePlaneNormal)
    reference5 = body.HybridShapes.Item(plane_name)
    sketch = body.sketches.add(reference5)`

You can code like above to create a 3D point, a normal, and refer it as your sketch plane.

evereux, 2022-08-18:
I would have and have done pretty much exactly the same. Thanks for the feedback!