evereux/pycatia #122: [BUG] Type annotations aren't working in python 3.6

deloarts, 2022-11-27:
**Describe the bug**
**pycatia** doesn't work with python 3.6, as stated in the readme or setup.py.
The reason are the type annotations, which are supported in python `>= 3.9`.
I've added a screenshot down below, which should explain the problem.

**To Reproduce**
Steps taken to reproduce the behavior:
- New venv with python 3.6
- Try test setup `pytest -vx`
- Setup will fail with error `ERROR tests/in/test_application.py - TypeError: 'type' object is not subscriptable`

**Screenshots**
[image]
Screenhot is take from file [drawing_dimension.py](https://github.com/evereux/pycatia/blob/b71b55fa00e26abdfdd4c2b63fec1a654c4165ff/pycatia/drafting_interfaces/drawing_dimension.py#L619).

**Desktop (please complete the following information):**
 - OS: Win11
 - V5-6R2017
 - pycatia development (latest commit [b71b55f](https://github.com/evereux/pycatia/commit/b71b55fa00e26abdfdd4c2b63fec1a654c4165ff))

**Additional context**
If I remember correctly, python 3.6 type annotations should be:

```python
from typing import Tuple
def foo(bar) -> Tuple[int, int]:
    ...
```

But I think updating the required python version to `>=3.9` would be easier :D

evereux, 2022-11-28:
I've completely missed the ball on this one! Thanks for reporting.

You are right ofcourse. I tried running the tests with 3.8, no go. Couldn't even find an official python 3.6 anymore. 3.9 is OK.

I'll make these changes.

deloarts, 2022-11-29:
Thanks for upgrading!

closed 2022-11-29 by deloarts