evereux/pycatia #124: [BUG] catia() produces "invalid class string" error

defaltus, 2022-12-08:
**Describe the bug**
catia() doesn't work, as stated in the readme. When running catia() after importing it from pycatia, the error is "invalid class string"

**To Reproduce**
- New Python Conda/Mamba environment
- Install pywin32 from conda-forge
- Install pycatia via pip
- Running CATIA in the Background
- Running catia() from e.g. iPython shell will fail

**Screenshot**
[image]

**Expected behavior**
It should "connect" to the running CATIA application

**Desktop (please complete the following information):**
 - OS: Windows 10
 - CATIA: V5-6R2021
 - pywin32: 305
 - pycatia: 0.5.7

evereux, 2022-12-08:
I've just done a fresh install using python (from https://www.python.org/), setting up a virtualenv and install pycatia via pip and there is no problem here.

I've seen cases where when CATIA V5 hasn't installed correctly and the COM interface wasn't registered properly (or perhaps something else borked it). I would try and re-install CATIA V5. If it isn't that it might be something peculiar with conda/mamba conda-forge (I'm not famliar with them).

```C:\Users\evereux\testing
λ cd pycatia\

C:\Users\evereux\testing\pycatia
λ python -m virtualenv env
created virtual environment CPython3.10.8.final.0-64 in 3698ms
  creator CPython3Windows(dest=C:\Users\evereux\testing\pycatia\env, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=C:\Users\evereux\AppData\Local\pypa\virtualenv)
    added seed packages: pip==22.3.1, setuptools==65.5.1, wheel==0.38.4
  activators BashActivator,BatchActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator

C:\Users\evereux\testing\pycatia
λ env\Scripts\activate

C:\Users\evereux\testing\pycatia
(env) λ pip install pycatia
Collecting pycatia
  Downloading pycatia-0.5.7-py3-none-any.whl (1.1 MB)
     ---------------------------------------- 1.1/1.1 MB 5.9 MB/s eta 0:00:00
Collecting pywin32>=224
  Using cached pywin32-305-cp310-cp310-win_amd64.whl (12.1 MB)
Installing collected packages: pywin32, pycatia
Successfully installed pycatia-0.5.7 pywin32-305

C:\Users\evereux\testing\pycatia
(env) λ python
Python 3.10.8 (tags/v3.10.8:aaaf517, Oct 11 2022, 16:50:30) [MSC v.1933 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> from pycatia import catia
>>> catia()
Application(name="CNEXT")
>>>
```

defaltus, 2022-12-08:
Reinstalling CATIA V5-6 fixed it for me, thanks!

closed 2022-12-08 by defaltus

evereux, 2022-12-08:
Great! Thanks for reporting.