evereux/pycatia #126: [BUG] Part selection/edit not working as intended

NielsMeinke, 2022-12-18:
**Describe the bug**
It is not possible to activate/edit a part within a product.
Using the given functions only highlights the entries in the file tree.

**To Reproduce**
Steps taken to reproduce the behavior:
Tree structure
```
BaseProduct
   |
   +---PartInstance (initially shown as product in pyCATIA)
   |    +CatPart file
   |
   +- Product
```

```python
from pycatia import catia

CATIA = Dispatch('CATIA.Application')
caa = catia()

documents = self.caa.documents
part_product = document.product.products[0] # select the PartInstance
part_doc = PartDocument(product.reference_product.parent.com_object) 
part = part_doc.part # get the actual CatPart object

# try to make the part the active tree object
selection.add(part)
```
`part.edition()` seems to not work on parts either

**Expected behavior**

I want to activate the Part, as if i right-clicked it in the tree view, then selected `xxx object -> Edit`.
However this snippet only highlights the part in the file tree, and does not allow certain calculations on the part.

**Desktop (please complete the following information):**
 - OS: Windows 10 1809
 - CATIA V5-6R2018
 - pycatia version 0.5.7

evereux, 2022-12-21:
I can't see any evidence of a bug here.

> CATIA = Dispatch('CATIA.Application')

Within the context of the short script you have provided this won't work. Is this necessary?

> documents = self.caa.documents

In python self refers current instanced version of the class so you wouldn't ever use self like this. Also, this won't work in the context of this script either.

> # try to make the part the active tree object
> selection.add(part)

All you are doing here is adding the "part" to the selection object. Do you know why you have done this?

> part.edition() seems to not work on parts either

I have no idea what part.edition() should do. Could you point me to some documentation that explains what this is? I can't find anything.

Here is a quick script accessing the first part in a product:

```from pycatia import catia
from pycatia.mec_mod_interfaces.part import Part
from pycatia.product_structure_interfaces.product_document import ProductDocument
from pycatia.product_structure_interfaces.product import Product

caa = catia()
documents = caa.documents

root_product_document = caa.active_document
root_product_document = ProductDocument(root_product_document.com_object)
root_product = root_product_document.product
sub_products = root_product.products
first_sub_product = Product(sub_products[0].com_object)
part = Part(first_sub_product.com_object)```

Take a look at a number of examples provided to get a better idea of how to get started with pycatia.

evereux, 2023-03-04:
Since there has been no feedback I am closing this issue.

closed 2023-03-04 by evereux

lee-qinghua, 2023-07-09:
caa = catia()
document = caa.active_document
product = document.product
product = Product(product.com_object)
products = product.products

first_sub_product = Product(products[0].com_object)
part = Part(first_sub_product.com_object)
print(part.hybrid_bodies)
print(part.bodies)

==================
error
Traceback (most recent call last):
  File "C:\Users\A\PycharmProjects\pythonProject1\myproduct.py", line 15, in <module>
    print(part.hybrid_bodies)
  File "D:\dev\anconda\lib\site-packages\pycatia\mec_mod_interfaces\part.py", line 297, in hybrid_bodies
    return HybridBodies(self.part.HybridBodies)
  File "D:\dev\anconda\lib\site-packages\win32com\client\dynamic.py", line 511, in __getattr__
    raise AttributeError("%s.%s" % (self._username_, attr))
AttributeError: <unknown>.HybridBodies