evereux/pycatia #127: Add  new type of documentation available for creation[FEATURE REQUEST]

Djey51, 2022-12-20:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is.

Hi,
Today with pycatia, I can only create the following type of document : "Drawing";"Product";"Part"
Could you add please "CATMaterial","catalog" and more please ?
The method is the same than for the existing one --> CATIA.Documents.Add("CATMaterial")

Today I obtain the following error 👍 
[image]

Thanks in advance

evereux, 2022-12-21:
It looks like you're running an older version of pycatia as there is also FunctionalSystem document. Still, that doesn't help with your feature request. 

I'll take a look at the CATMaterial as the workbench is already in place to do that. 

Anything else is a little more time consuming as I would need to implement the workbenches so you could actually do something with that document type.

> Could you add please "CATMaterial","catalog" and more please ?

You're going to need to be more specific. Implementing everything would be quite a lot of work and the reason it's not already there. Also, I'm limited on access to licenses so in some cases will be just me guessing as to whether something works or not.

evereux, 2022-12-21:
Can you provide a simple sample CATScript / VBA for catalog please? 

All I've got is the `[CatalogDocument](http://catiadoc.free.fr/online/CAAScdBase/CAAScdAutomationHome.htm)` class that has three methods. That doesn't seem all that useful to me so I think I'm missing more functionality somewhere.

Djey51, 2022-12-21:
Hi @evereux,
Effectively, it seems that you don'y have enough CatiaV5 documentation.
As I'm a Catia administrator and expert for a big company, I'll provide you the complete API for catia automation if you want or just the part concerned by my requests if you prefere (but I think it is available on the net [Root page](http://catiadoc.free.fr/online/CAAScdBase/CAAScdAutomationHome.htm)  and  [Here](http://catiadoc.free.fr/online/CAAScdBase/CAAScdAutomationHome.htm) for the matetial)
For now, i have done my macro through BVA for Catia for the CATMaterial so it is not early needed. But for the future, I'm very entousiat to be able to manage all my macro only with python. It should be very powerfull.
If you need help with Catia documentation, don't hesitate to ask me.
Best regards

evereux, 2022-12-22:
The latest I have is the `V5Automation.chm` from CATIA V5 R28. Any documentation you have that expands on what I have would be great but I'm not sure that's the issue here? I just don't know what modules I need to import to expand on the catalog functionality as the `V5Automation.chm` doesn't give me any clues.

Your links only point me to the home page of that site (the same happened with my link above I now realise) which is pretty much a carbon copy of the `V5Automation.chm` content wise. I'm sure they're generated from the same source. This is the [CatalogDocument](http://catiadoc.free.fr/online/interfaces/interface_CatalogDocument.htm) I meant to link to.

Djey51, 2022-12-22:
Hi @evereux,
You are right about catalog, and sorry for that because i didn't think before i wrote. And in fact, there is a lack on catalog component with Catia automation... And only what you have in your link is possible to do with a macro. Everything else is missing in API. So i said catalog but it was not the good exemple. In fact I wanted to ask to be able to create more type of file and especially the catmaterial file (but aldo, .stp, .cgr,.cgm, and so on)

evereux, 2022-12-24:
> In fact I wanted to ask to be able to create more type of file and especially the catmaterial file (but aldo, .stp, .cgr,.cgm, and so on)

You are talking about quite different things here. CATMaterial / Catalog files are native CATIA V5 files and are related to the original request `... CATIA.Documents.Add("CATMaterial") ...`. 

As far as I know `.stp`, `.cgr` and `.cgm` files are only created by using `save as` and there are examples showing how you would do that. You wouldn't create these file types using `Documents.Add()`.

I don't know what an `aldo` file is.

evereux, 2022-12-24:
Example of CGR: https://github.com/evereux/pycatia/blob/master/examples/example__document__003.py

evereux, 2023-03-04:
I'm closing this thread as it's starting to cover many areas which will make this hard to follow in the long run.

If you have a specifc request please raise a new github issue.

closed 2023-03-04 by evereux