evereux/pycatia #128: [BUG] Add a new dimension into  dimensions fail on drawing 2d

lampt2010, 2022-12-25:
**Describe the bug**
I used to **add2** function from class **DrawingDimensions** . I watched the debugger and check results on catia and saw nothing happened . The code didn't have an exception but the results were difference my forecast .
Below is my code and results 

`def get_background_view_and_factory(sheet: DrawingSheet) :

    views = sheet.views
    main_view = DrawingView( views.get_item_by_name('Main View'))
    background_view = DrawingView(views.get_item_by_name('Background View').com_object)
    background_view.activate()
    factory_2d =Factory2D(background_view.factory_2d.com_object)

    point_a = ( factory_2d.create_point(30, 40))
    point_b =  factory_2d.create_point(50, 60)
    line_ab = factory_2d.create_line(point_a.get_coordinates()[0],point_a.get_coordinates()[1],point_b.get_coordinates()[0],point_b.get_coordinates()[1])
    geomatre_element = line_ab.geometric_element
    dimensions = DrawingDimensions( background_view.dimensions.com_object)
    # add new dimension between 2 points a b
    dimension = dimensions.add2(0, [point_a.get_coordinates(),point_b.get_coordinates()],[point_a.get_coordinates(),point_b.get_coordinates()],line_ab,0)`

**Screenshots**

[image]

[image]

**Desktop (please complete the following information):**
 - OS: Microsoft Windows 10 Enterprise ( 10.0.19045 N/A Build 19045)
 - CATIA V5 version V5R21
 - pycatia version : 0.5.7

**Additional context**
according to my opinion about reading the document pycatia , results must return collection dimensions including 1 item .
Someone can explain to me

evereux, 2022-12-28:
This isn't a bug. You need to pass the com_object of the dimension objects to the dimensions.add2() method.

Also, there were a number of errors in your sample script.

Here is a working example using your script as the base.

```python
from pycatia import catia
from pycatia.drafting_interfaces.drawing_dimensions import DrawingDimensions
from pycatia.drafting_interfaces.drawing_document import DrawingDocument
from pycatia.drafting_interfaces.drawing_view import DrawingView
from pycatia.sketcher_interfaces.factory_2D import Factory2D

caa = catia()
drawing_document = DrawingDocument(caa.active_document.com_object)
sheets = drawing_document.sheets
sheet = sheets[0]

views = sheet.views
main_view = DrawingView(views.get_item_by_name('Main View').com_object)
background_view = DrawingView(views.get_item_by_name('Background View').com_object)
background_view.activate()
factory_2d = Factory2D(background_view.factory_2d.com_object)

point_a = factory_2d.create_point(30, 40)
point_b = factory_2d.create_point(50, 60)
line_ab = factory_2d.create_line(point_a.get_coordinates()[0], point_a.get_coordinates()[1], point_b.get_coordinates()[0], point_b.get_coordinates()[1])
geometric_element = line_ab.geometric_element
dimensions = DrawingDimensions(background_view.dimensions.com_object)
# add new dimension between 2 points a b
dimension = dimensions.add2(0, [point_a.com_object, point_b.com_object], [point_a.com_object, point_b.com_object], line_ab.com_object, 0)
```

lampt2010, 2022-12-28:
> This isn't a bug. You need to pass the com_object of the dimension objects to the dimensions.add2() method.
> 
> Also, there were a number of errors in your sample script.
> 
> Here is a working example using your script as the base with the errors fixed.
> 
> ```python
> from pycatia import catia
> from pycatia.drafting_interfaces.drawing_dimensions import DrawingDimensions
> from pycatia.drafting_interfaces.drawing_document import DrawingDocument
> from pycatia.drafting_interfaces.drawing_view import DrawingView
> from pycatia.sketcher_interfaces.factory_2D import Factory2D
> 
> caa = catia()
> drawing_document = DrawingDocument(caa.active_document.com_object)
> sheets = drawing_document.sheets
> sheet = sheets[0]
> 
> views = sheet.views
> main_view = DrawingView(views.get_item_by_name('Main View').com_object)
> background_view = DrawingView(views.get_item_by_name('Background View').com_object)
> background_view.activate()
> factory_2d = Factory2D(background_view.factory_2d.com_object)
> 
> point_a = factory_2d.create_point(30, 40)
> point_b = factory_2d.create_point(50, 60)
> line_ab = factory_2d.create_line(point_a.get_coordinates()[0], point_a.get_coordinates()[1], point_b.get_coordinates()[0], point_b.get_coordinates()[1])
> geometric_element = line_ab.geometric_element
> dimensions = DrawingDimensions(background_view.dimensions.com_object)
> # add new dimension between 2 points a b
> dimension = dimensions.add2(0, [point_a.com_object, point_b.com_object], [point_a.com_object, point_b.com_object], line_ab.com_object, 0)
> ```

evereux, 2022-12-28:
Closing as issue is resolved.

closed 2022-12-28 by evereux

evereux, 2022-12-28:
Just for your information I may change a number of these methods so you don't need to pass the com_object. I'll bump the version up to 0.6.x for that though as this will likely be a breaking change for people.