evereux/pycatia #130: [FEATURE REQUEST]"Nothing" object

alangch, 2023-02-23:
Sometimes, we need to use a "Nothing" object in a parameter position.
Such as creating a sphere, if we just want to use the absolute axis system, in VBA, we can write like this

set sphere = HybridShapeFactory.AddNewSphere(centerPt, Radius, Nothing, -45, 45, 0, 180)

So in pycatia, what should I write at the position of the "Nothing" parameter?
I tried to ues None but it didn't work.

evereux, 2023-02-23:
This error has me a bit lost as I don't know how to fix this short of 
wrapping the contents of class method hybrid_shape_factory.add_new_sphere()
within a vba function. This shouldn't be that big a deal but it would be nice 
to know everywhere else this is a potential issue.

For reference I did find this on stackoverflow [Pass the variable 
`Nothing` using Python's  `win32com`](https://stackoverflow.com/questions/41175007/pass-the-variable-nothing-using-pythons-win32com)
but I couldn't get these solutions to work. Here is my sample script:

```python
from pycatia import catia
from pycatia.mec_mod_interfaces.part import Part
from pycatia.hybrid_shape_interfaces.hybrid_shape_factory import HybridShapeFactory
import pythoncom

caa = catia()
document = caa.active_document
part = Part(document.part.com_object)

hybrid_bodies = part.hybrid_bodies
construction_geometry = hybrid_bodies.item("ConstructionGeometry")

hybrid_shapes = construction_geometry.hybrid_shapes
centre_point = hybrid_shapes.item("Point.1")
reference_centre_point = part.create_reference_from_object(centre_point)

vba_nothing = pythoncom.Nothing

hybrid_shape_factory = part.hybrid_shape_factory
hybrid_shape_sphere = hybrid_shape_factory.add_new_sphere(reference_centre_point, vba_nothing, 100, -45, 45, 0, 180)
```

In [hybrid_shape_factory.add_new_sphere](https://github.com/evereux/pycatia/blob/master/pycatia/hybrid_shape_interfaces/hybrid_shape_factory.py#L4890) I did also try replacing `i_axis.com_object` with just `i_axis` with respect to this test.

ptm-tm, 2023-03-07:
if u want ti fill all parameters in sweep
`   vba_nothing = pythoncom.Nothing 
    Wall=hsf.add_new_sweep_line(Profile_Pad)
    Wall.mode=6
    Wall.solution_no=0
    Wall.smooth_activity=False
    Wall.guide_deviation_activity=False
    Wall.draft_computation_mode=0
    Wall.draft_direction=Hybrid_Shape_D3
    Wall.set_first_length_definition_type(3,part_document.create_reference_from_object(Plane_Zmax_offset))
    Wall.set_second_length_definition_type(3,vba_nothing)
    Wall.setback_value=0.02    
    Wall.fill_twisted_areas = 1
    Wall.c0_vertices_mode = True
    Wall.append_hybrid_shape(Profile_Pad)
    Wall.name="Wall_sweep"
    hybridBody2.append_hybrid_shape(Wall)`
    Wall.set_second_length_definition_type(3,vba_nothing)

evereux, 2023-09-15:
I have added a comment to the docstring regarding this issue. 

See the following commit: https://github.com/evereux/pycatia/commit/9d43165aab013f073ee4a44a0ed09baa6edf3397

evereux, 2023-10-05:
I've got this fixed in the development branch now. see [example__hybrid_shape_factory__006](https://github.com/evereux/pycatia/blob/developement/examples/example__hybrid_shape_factory__006.py) for example usage.

closed 2023-10-05 by evereux