evereux/pycatia #137: [FEATURE REQUEST]

ptm-tm, 2023-03-02:
**Is your feature request related to the CAA V5 Automation API?**
If yes, please describe where in the API documentation that feature is described.

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

need add property
part.hybrid_shape_factory
part.hybrid_bodies
document.part

autocomplete is not worked(
[image]

deloarts, 2023-03-03:
See [#107](https://github.com/evereux/pycatia/issues/107#issuecomment-1336195688)

ptm-tm, 2023-03-03:
yes its the same)

closed 2023-03-03 by ptm-tm