evereux/pycatia #138: [BUG]

yassinedjemel, 2023-03-03:
**Describe the bug**
Missing attribute in 'HybridShapeFactory' object

**Expected behavior**
I wanted to get the dimensions of the box that my part is contained in.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - CATIA V5 version 5.21
 - pycatia version e.g. pycatia 0.5.7

I wrote this code to extract the dimensions of the box that my catia part is contained in but i always keep on getting the error: 'HybridShapeFactory' object has no attribute 'add_new_box'

from pycatia import catia

caa = catia()
documents = caa.documents
document = caa.active_document
part = document.part
part.update()

# Get the main body of the part
body = part.main_body

# Get the HybridShapeFactory object
hybrid_shape_factory = part.hybrid_shape_factory

# Create a box that encompasses the entire part
bounding_box = hybrid_shape_factory.add_new_box(body.bounding_box().get_min_corner(), body.bounding_box().get_max_corner())

# Get the length, width, and height of the bounding box
length = bounding_box.get_length()
width = bounding_box.get_breadth()
height = bounding_box.get_height()

print(f"Length: {length}")
print(f"Width: {width}")
print(f"Height: {height}")

evereux, 2023-03-04:
I have searched the CAA V5 Visual Basic Help for "AddNewBox" and it isn't returning anything. I also checked the HybridShapeFactory Interface Index and read through all the methods and I couldn't see anything,

Can you point me to some documentation on this please?

ptm-tm, 2023-03-04:
hm
[FillingSurfaces.zip](https://github.com/evereux/pycatia/files/10888195/FillingSurfaces.zip)
m why all need bounding box? catia has own command c:creates rough stock or u can use standart UDF from catia catalog (this V5R19).... and you can use ton of vb scripts for bounding box... u relly need python script for this?

HybridShapeFactory  dont have metods or properties thats describe bounding box.
V5automation can only show bounding box (Property BoundingBoxSelectionMode( ) As boolean)

evereux, 2023-03-04:
>  catia has own command c:creates rough stock

Just for clarity for people reading this this feature is license dependent. This is why people use scripting I suspect.

> and you can use ton of vb scripts for bounding box... u relly need python script for this?

There's a ton of scripts already written in vb that I'm sure people are writing in python using pycatia because they prefer to do so.

It would be nice to have a version written in python using pycatia and added as an example. Nice challenge for someone? It's a not a straight forward task to do properly for complex shapes with no reference axis.

ptm-tm, 2023-03-04:
To write this, i need to know what he want receive. 
For people who don't have license for ` c:creates rough stock` , catia have udf, but may be it need another license

ptm-tm, 2023-03-04:
and `c:bounding box` for finally

evereux, 2023-10-05:
Closing this as original question isn't supported by CATIA V5 out of the box from what I can tell.

closed 2023-10-05 by evereux

ptm-tm, 2023-10-05:
im close to write script to create bounding box)))