evereux/pycatia #141: [BUG]

ptm-tm, 2023-03-07:
**Describe the bug**
A clear and concise description of what the bug is.
Cant set limits for extrude

**To Reproduce**

    Wall=hsf.add_new_extrude(Profile_Pad,0,10,Hybrid_Shape_D3)
    Wall.context=0
    Wall.first_limit_type=2
    Wall.second_limit_type=2
    #Wall.first_upto_element(Plane_Zmax)
    Wall.second_upto_element(ref_XY)
    Wall.name="Wall"
    hybridBody2.append_hybrid_shape(Wall)

where hsf=hybrid_shape_factory
Profile pad= join 4 closed line perpendicullar to Hybrid_shape_dir
Axis_Coord=Axis_System.get_x_axis()
Hybrid_Shape_D1=hsf.add_new_direction_by_coord(Axis_Coord[0], Axis_Coord[1], Axis_Coord[2]) 
ref_XY-reference of plane

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - Windows 11
 - CATIA V5 B29
 - pycatia 0.5.7

When i delete sets limits- extrude works.

**Additional context**
*\examples\example__Bounding_box__001.py", line 475, in <module>
    Wall.second_upto_element(ref_XY)
    ^^^^^^^^^^^^^^^^^^^^^^^^
*\PyCatia\Lib\site-packages\pycatia\hybrid_shape_interfaces\hybrid_shape_extrude.py", line 404, in second_upto_element
    return Reference(self.hybrid_shape_extrude.SecondUptoElement)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  *PyCatia\Lib\site-packages\win32com\client\dynamic.py", line 628, in __getattr__
    ret = self._oleobj_.Invoke(retEntry.dispid, 0, invoke_type, 1)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
pywintypes.com_error: (-2147352567, 'Ошибка.', (0, 'CATIAHybridShapeExtrude', 'The method SecondUptoElement failed', None, 0, -2147467259), None)

evereux, 2023-03-07:
Can you 

* post a complete (stripped down to just this issue) example please? Not just the bits you *think* I need to see. 😄 

I have a number of questions which would be answered if I had that.

ptm-tm, 2023-03-07:
like this?
#https://github.com/ptm-tm/pycatia/commit/9bcd4fe8419b6f66c5b2bc79c448ed4e4c79ed75

ptm-tm, 2023-03-07:
not matter what in this field
`Wall.second_upto_element(part_document.create_reference_from_object(Plane_Zmax_offset))`
or
`Wall.second_upto_element(refXY)`

evereux, 2023-03-07:
What I need is code that I can run that deals with just the hsf.add_new_extrude() method.

I don't want to pick through code that's not relevant to just that. Saves a lot of time.

ptm-tm, 2023-03-07:
this full code
https://github.com/ptm-tm/pycatia/blob/bounding-box/examples/example__Bounding_box__001.py

evereux, 2023-03-07:
Please read what I have posted.

ptm-tm, 2023-03-07:
[test extrude.zip](https://github.com/evereux/pycatia/files/10908700/test.extrude.zip)
max clear

evereux, 2023-03-07:
>not matter what in this field
> Wall.second_upto_element(part_document.create_reference_from_object(Plane_Zmax_offset))
> or
> Wall.second_upto_element(refXY)

> max clear

I don't what you mean. 😄 

Here's what I expect from a typical bug report:

# [BUG] Can't extrude profile with limits

The code example below fails when creating an extrusion. To run the code you will need:

* a geometrical set called `ContructionGeometry`
* two limiting planes, `Plane.1` and `Plane.2`

```python
from pycatia import catia
from pycatia.mec_mod_interfaces.part import Part
from pycatia.hybrid_shape_interfaces.hybrid_shape_plane_explicit import HybridShapePlaneExplicit
from pycatia.mec_mod_interfaces.hybrid_body import HybridBody

caa = catia()
document = caa.active_document
part = Part(document.part.com_object)

hsf = part.hybrid_shape_factory
origin_elements = part.origin_elements

hsp_explicit_1 =  HybridShapePlaneExplicit(origin_elements.plane_yz.com_object)
reference_hsp_explicit_1 = part.create_reference_from_object(hsp_explicit_1)

hs_direction_1 = hsf.add_new_direction(reference_hsp_explicit_1)

hybrid_bodies = part.hybrid_bodies
construction_geometry = HybridBody(hybrid_bodies.get_item('ConstructionGeometry').com_object)
hybrid_shapes = construction_geometry.hybrid_shapes

sketches = construction_geometry.hybrid_sketches
sketch_1 = sketches.item('Sketch.1')

reference_sketch = part.create_reference_from_object(sketch_1)

hs_extrude_1 = hsf.add_new_extrude(reference_sketch, 20, 0, hs_direction_1)
hs_extrude_1.first_limit_type = 2
hs_extrude_1.second_limit_type = 2

hs_plane_offset_1 = hybrid_shapes.item('Plane.1')
hs_plane_offset_2 = hybrid_shapes.item('Plane.2')

reference_hs_plane_offset_1 = part.create_reference_from_object(hs_plane_offset_1)
reference_hs_plane_offset_2 = part.create_reference_from_object(hs_plane_offset_2)

hs_extrude_1.first_upto_element = reference_hs_plane_offset_1
hs_extrude_1.second_upto_element = reference_hs_plane_offset_2

hs_extrude_1.symmetrical_extension = 0

construction_geometry.append_hybrid_shape(hs_extrude_1)

part.update()
```

This is the following traceback:

```Traceback (most recent call last):
  File "C:\Users\evereux\github\pycatia\testing.py", line 39, in <module>
    hs_extrude_1.first_upto_element = reference_hs_plane_offset_1
  File "C:\Users\evereux\github\pycatia\pycatia\hybrid_shape_interfaces\hybrid_shape_extrude.py", line 306, in first_upto_element
    self.hybrid_shape_extrude.FirstUptoElement = reference_element
  File "C:\Users\evereux\github\pycatia\env\lib\site-packages\win32com\client\dynamic.py", line 686, in __setattr__
    self._oleobj_.Invoke(entry.dispid, 0, invoke_type, 0, value)
TypeError: must be real number, not Reference
```

# Solution

`pycatia\hybrid_shape_interfaces\hybrid_shape_extrude.py`

```python
    @first_upto_element.setter
    def first_upto_element(self, reference_element: Reference):
        """
        :param Reference reference_element:
        """

        self.hybrid_shape_extrude.FirstUptoElement = reference_element.com_object
```

should be

```    @first_upto_element.setter
    def first_upto_element(self, reference_element: Reference):
        """
        :param Reference reference_element:
        """

        self.hybrid_shape_extrude.FirstUptoElement = reference_element.com_object.com_object
```

I'll submit a patch now.

ptm-tm, 2023-03-07:
sry im not programmer)  im write fast code to test it.
and i cant find solution :( if i find i push it

ptm-tm, 2023-03-07:
code need only axis system ( but in many cases axis system create automatically when start part documetn.
and i dont understand how u colored code

evereux, 2023-03-07:
> sry im not programmer) im write fast code to test it.
> and i cant find solution :( if i find i push it

I didn't expect the solution. Just a description of the issue I can understand and code I can run.

> code need only axis system ( but in many cases axis system create automatically when start part documetn.

I don't understand to what you are referring.

> and i dont understand how u colored code

[Raising issues / asking questions](https://github.com/evereux/pycatia/issues/28)

evereux, 2023-09-11:
As far as I know the reported issue has been fixed.

closed 2023-09-11 by evereux