evereux/pycatia #153: [BUG] Group object does not have attribute add_explicit

Raph-xyz, 2023-08-08:
**Describe the bug**
When using the (documented) attribute add_explicit, an AttributeError is raised
`testGroup.add_explicit(partInstance)`

Error is : 
```
Exception has occurred: AttributeError
<unknown>.add_explicit
  File "C:\Users\RaphaelBIET\repo\c-checker\coordinator\groupIssue.py", line 31, in main
    testGroup.add_explicit(partInstance)
    ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\RaphaelBIET\repo\c-checker\coordinator\groupIssue.py", line 53, in <module>
    main()
AttributeError: <unknown>.add_explicit
```
**To Reproduce**
I simplified some code as much as I could to get this.

```python
import pycatia
import os
import platform

def main():
    print(f"PyCatia version is {pycatia.__version__}")
    print(f"OS is {platform.system()} rel {platform.release()} and version {platform.version()}")

    try:
        caa = pycatia.catia()
    except:
        print("Catia is not open, please open Catia before running this script")
        return

    print("Add a Product")
    caa.documents.add("Product")
    topNode = caa.active_document.product

    print("Add groups")
    groups = topNode.get_technological_object("Groups")
    testGroup = groups.add()
    testGroup.name = "test"

    # part is attached to the issue, place accordingly in folders
    cadsPath = os.path.abspath("./examples/")
    partPath = os.path.join(cadsPath, "CUBE.CATPart")
    matrix = (1.000, 0, 0, 0, 0.707, 0.707, 0, -0.707, 0.707, 10, 11, 12)
    partInstance = insertPartMove(partPath, topNode, matrix)

    # Error Here:
    testGroup.add_explicit(partInstance)

def insertPartMove(partPath, rootNode, matrix):
    print("insert")
    # add to root
    rootNode = pycatia.product_structure_interfaces.product.Product(rootNode.com_object)
    iItem = rootNode.products.count
    rootNode.products.add_components_from_files(("", partPath), "All")

    # 'if part is well added, then move it
    if iItem < rootNode.products.count:
        childInstance = rootNode.products[rootNode.products.count - 1]
    else:
        print("The part is not added to the product : " + partPath)
        return

    # position
    childInstance.move.apply(matrix)
    return childInstance

main()
```

**Expected behavior**
The part is added to the Catia group (when selecting the group, the contained part is also selected)

before: 
[image]

after: 
[image]

**Desktop:**
PyCatia version is 0.5.8
OS is Windows rel 10 and version 10.0.22621
CATIA V5-6R2017

**Additional context**
I have studied the problem with some people proficient in Python, and I am quite used to Catia so I'm confident there is a pycatia issue. With Catia object model I can never be sure, but I did try a lot of things!

evereux, 2023-08-09:
I couldn't get your code to work as there were errors elsewhere. Also, it still could do with stripping down more to debug just this error.

However, I wrote this and it seems to work for me:

```python

from pathlib import Path

from pycatia import catia
from pycatia.navigator_interfaces.groups import Groups
from pycatia.product_structure_interfaces.product import Product

cad_file = Path('C:/Users/evereux/cloud/pycatia/tests/cat_files/part_measurable.CATPart')
file_list = ("", str(cad_file))

caa = catia()
documents = caa.documents
document = caa.active_document
product = document.product
product = Product(product.com_object)
groups = product.get_technological_object("Groups")
groups = Groups(groups)
test_group = groups.add()
test_group.name = "test"

# new_document = documents.open(cad_file)
# new_product = new_document.product
product.products.add_components_from_files(file_list, "All")
products = product.products
product_just_added = products.item(len(products))

test_group.add_explicit(product_just_added)
```

evereux, 2023-08-09:
btw thanks for the time and effort in making this bug report the way you did. it helped me a great deal.

evereux, 2023-09-11:
I can't reproduce the bug and my example works as expected so closing due to no feedback.

closed 2023-09-11 by evereux