evereux/pycatia #154: [FEATURE REQUEST] [INFORMATION] Method to find Product Author - Creation Date

Mummuu3i1, 2023-09-05:
Hello there,

First of all, thank you for this library which allows us to automate tasks via python.

Let me explain my request / problem: 

I'm a design teacher. My students have work to do on CATIA and to hand in to me. 

I'd like to be able to query these jobs to retrieve the following information: 

1. Author
2. Original creation date

We can find this information in the "Document Properties" section directly in CATIA. 

However, I'd like to create a routine to query the part. 

I've looked through the pycatia documentation, but haven't found anything for this purpose. (No method with a Get_Info etc.)

@evereux  do you know if we have another way of doing this? If not, do you think you could add a method with this kind of attributes ?

evereux, 2023-09-05:
I think you're talking about a user defined property?

This should help point you in the right direction.

```from pycatia import catia
from pycatia.product_structure_interfaces.product import Product

caa = catia()
document = caa.active_document
product = Product(document.product.com_object)
user_def_properties = product.user_ref_properties

for property in user_def_properties:
    print(property.name)
    print(property.value_as_string())
```

evereux, 2023-09-05:
ps if this isn't what you're looking for please provide a screenshot.

Mummuu3i1, 2023-09-05:
Hi evereux,

Thank for you feedback. 
These are the properties I'm looking for. 
However, I've found that if we don't enter them manually, nothing appears in my promp after running the script. So I have to look at the file's metadata and not the part properties. 

Thanks for your help.

evereux, 2023-09-05:
> However, I've found that if we don't enter them manually, nothing appears in my promp after running the script. 

I don't follow. How else are they entered? Can you explain your workflow please?

> So I have to look at the file's metadata and not the part properties.

That's not something that's accessible with the CATIA COM object as far as I know? So will be out of the scope of this project. Shouldn't be too hard to knock up a script that does this yourself though if it's stored in a human readable format within the file.

Mummuu3i1, 2023-09-05:
I need to know if my students are cheating in CATIA tests. 
To do this, I wanted to create a python script and retrieve information such as the file author and the creation date directly from the part properties. 

However, I realised that the author and creation data are custom properties. So they have to enter this information manually. There are no "solutions" for this information to be generated automatically.

Translated with www.DeepL.com/Translator (free version)

Mummuu3i1, 2023-09-05:
I need to know if my students are cheating in CATIA tests. 
To do this, I wanted to create a python script and retrieve information such as the file author and the creation date directly from the part properties. 

However, I realised that the author and creation data are custom properties. So they have to enter this information manually. There are no "solutions" for this information to be generated automatically.

evereux, 2023-09-05:
So, you want to ensure that the student is creating a new file each time and not editing someone else's to make it look like they've done the work themselves?

I'm still unclear on the work flow though. Can you get a screenshot of these custom properties that shows how they're added / how you check them manually? Do blank out any personal information if needed.

evereux, 2023-09-11:
Lacking information from OP about what it is they're trying to do exactly.

Please feel free to re-open if a full explanation can be provided.