evereux/pycatia #155: [FEATURE REQUEST]

hosseintfr, 2023-09-29:
How can add a constraint (like coincidence between two planes of two cubes) in a product by pycatia?
Please give me an example code. I have read the Constraints example but it isn't enough to learn how to define a constraint between two objects. Constraints example just is for fixing one part in a product and not two elements of two different parts in the product.
Thanks

hosseintfr, 2023-09-30:
@evereux Can you please help me? I will be very grateful if you help me.

evereux, 2023-09-30:
> Constraints example just is for fixing one part in a product and not two elements of two different parts in a product.

The examples aren't ever going to provide solutions to all problems I'm afraid. But, working through all the examples should at least provide an improved understanding of how to use pycatia.

I can't solve this problem for you for the time being but what I would do is:

Record a CATIA macro doing what it is that you want to achieve (**carrying out those steps only**). Then, look at the script generated. This will give you a starting point for you to be able to rewrite the code in python using pycatia.

If you still can't solve this issue please supply the script generated by CATIA and also the python script you have created.

hosseintfr, 2023-09-30:
thanks. my python code is:
from pycatia import catia
from pycatia.product_structure_interfaces.product_document import ProductDocument
from pycatia.mec_mod_interfaces.part_document import PartDocument
from pycatia.mec_mod_interfaces.constraints import Constraints
from pycatia.mec_mod_interfaces.constraints import Constraint
from pycatia.enumeration.enumeration_types import cat_constraint_type
from pycatia.in_interfaces.reference import Reference
from pycatia.mec_mod_interfaces.part import Part
from pycatia.in_interfaces import viewer

caa = catia()
documents = caa.documents

documents.open(r"C:\Users\u6790\Desktop\catiaTR\Part1.CATPart")
part_doc = PartDocument(caa.active_document.com_object)
part_product = part_doc.product
part_product.part_number = "Part_one"

documents.open(r"C:\Users\u6790\Desktop\catiaTR\Part2.CATPart")
part_doc2 = PartDocument(caa.active_document.com_object)
part_product2 = part_doc2.product
part_product2.part_number = "Part_two"

documents.add('Product')
product_doc = ProductDocument(caa.active_document.com_object)
product = product_doc.product
product.part_number = "Assembly"

product.products.add_component(part_product)
product.products.add_component(part_product2)
product.update()
part_doc.close()
part_doc2.close()

caa.active_window.active_viewer.reframe()

#################### Constraints ###################

part1 = Part(part_doc.part.com_object)
part2 = Part(part_doc2.part.com_object)

a1 = part1.origin_elements.plane_zx
b1 = Reference(a1)

a2 = part2.origin_elements.plane_xy
b2 = Reference(a2)

__________________________________________________________
And my Macro code is:

Language="VBSCRIPT"

Sub CATMain()

Dim productDocument1 As Document
Set productDocument1 = CATIA.ActiveDocument

Dim product1 As Product
Set product1 = productDocument1.Product

Dim constraints1 As Collection
Set constraints1 = product1.Connections("CATIAConstraints")

Dim reference1 As Reference
Set reference1 = product1.CreateReferenceFromName("Assembly/Part_one.1/!Selection_RSur:(Face:(Brp:(Pad.1;0:(Brp:(Sketch.1;4)));None:();Cf11:());Pad.1_ResultOUT;Z0;G8252)")

Dim reference2 As Reference
Set reference2 = product1.CreateReferenceFromName("Assembly/Part_two.1/!Selection_RSur:(Face:(Brp:(Pad.1;2);None:();Cf11:());Pad.1_ResultOUT;Z0;G8252)")

Dim constraint1 As Constraint
Set constraint1 = constraints1.AddBiEltCst(catCstTypeSurfContact, reference1, reference2)

product1.Update 

End Sub

hosseintfr, 2023-09-30:
@evereux Thanks a lot. I tried to add a contact constraint by reading the macro. It was very helpful and I add these lines to my code and it worked:

reference1 = product.create_reference_from_name("Assembly/Part_one.1/!Selection_RSur:(Face:(Brp:(Pad.1;0:(Brp:(Sketch.1;4)));None:();Cf11:());Pad.1_ResultOUT;Z0;G8252)")
reference2 = product.create_reference_from_name("Assembly/Part_two.1/!Selection_RSur:(Face:(Brp:(Pad.1;2);None:();Cf11:());Pad.1_ResultOUT;Z0;G8252)")
constraint1 = constraints.add_bi_elt_cst(cat_constraint_type.index("catCstTypeSurfContact"), reference1, reference2)

hosseintfr, 2023-09-30:
For who read this issue, I should say that, if you want to add a constraint in your product, first start recording a macro, then define the constraint and stop recording. Then open the macro and copy the lines about defining the constraint. Just change some of commands to match with pycatia commands. For example, instead of "CreateReferenceFromName", use "create_reference_from_name". Also use "add_bi_elt_cst" instead of "AddBiEltCst". 
For defining the references of your constraint, use the names that are in your macro script.

evereux, 2023-10-02:
Great stuff. Glad you managed to figure it out. 👍

evereux, 2023-10-05:
Question appears to be resolved.

closed 2023-10-05 by evereux