evereux/pycatia #163: [FEATURE REQUEST] add GetCustomFactory methode

Djey51, 2023-10-10:
**Is your feature request related to the CAA V5 Automation API?**
If yes, please describe where in the API documentation that feature is described.

[image]
[image]

They are also examples in Catia api doc.

I need this methode to be able to use the power copy which is  a very powerfull feature in CatiaV5.
I have done it thanks to catvba but I can't reproduce it with pycatia due to this lack of methode.

For powercopy, it is the InstanceFactory to be use with the GetCustomerFactory methode.

Thanks

evereux, 2023-10-10:
That's already there? 

https://github.com/evereux/pycatia/blob/51503e328a521a7be5ab143976fd458988d09f5c/pycatia/mec_mod_interfaces/part.py#L766

Djey51, 2023-10-10:
Usually, I declare part object as following:

part_document = caa.active_document
part =part_document.Part
Then if I write "part.", I don't have autocompletion proposed

Do you think I should declare my part object as followind instead?

part_document = caa.active_document
part = Part(part_document.part.com_object) 

Thanks for your answer @evereux

deloarts, 2023-10-10:
> I have auto completion for other objects but not for caa.active_document.part object.

See https://github.com/evereux/pycatia/blob/master/examples/example__hybrid_bodies__001.py#L37

Or #107

Djey51, 2023-10-10:
Thank you very much for your answer @deloarts 

It is more clear for me now.
I'll test it tomorrow and close the issue.
Best regards

Mithro86, 2023-10-10:
@Djey51 Some copy and paste from my own project. Hope it can be of help. 

```import os
import sys

sys.path.insert(0, os.path.abspath('..\\pycatia'))

from pycatia import catia
from pycatia.mec_mod_interfaces.instance_factory import InstanceFactory

caa = catia()
document = caa.active_document
part = document.part
factory = part.get_customer_factory("InstanceFactory")
instance_factory = InstanceFactory(factory.com_object)

# close factory if already open
instance_factory.end_instance_factory() 

# start instance factory and get power copy
instance_factory.begin_instance_factory("Power_copy_name", my_absolute_dirpath + "\\master.CATPart")
instance_factory.instantiation_mode = "After"
instance_factory.begin_instantiate()
instance_factory.put_input_data("Input_1", reference_1)
instance_factory.put_input_data("Input_2", reference_2)
instance_factory.instantiate()
instance_factory.end_instantiate()
instance_factory.end_instance_factory()
```

Djey51, 2023-10-11:
Hi all,

Thank you very much for your help.
Now it works perfectly!!

Thanks @Mithro86 for your exemple which make me earn my time.

I have to reconsidere my way of declaring objects for y futur developments.

Thanks to give me the good direction.

Have a good day and long life to pycatia.
(End life of Catia V5 is about 2030)

closed 2023-10-11 by Djey51

evereux, 2023-10-11:
Thanks for the kind words @Djey51 and thanks to both @Mithro86 and @deloarts for helping.