evereux/pycatia #164: Creating development and production environments (step by step)

evereux, 2023-10-11:
## Development and production environments.

How to make seperate environments for project and development purposes.

### The base folder in your home directory. Please replace `~` with 
path you wish to use.

```console
cd ~
mkdir pycatia-projects
cd .\pycatia-projects\
```

### github (development)

```console
cd .\pycatia-projects\
# deactivate any virtualenv if active
deactivate
# clone the github repo
git clone https://github.com/evereux/pycatia.git pycatia-git
cd pycatia-git
python -m virtualenv env
.\env\Scripts\activate
pip install -r .\requirements\requirements.txt
```

### pypi (production)

```console
# deactivate any virtualenv if active
deactivate
cd ~\pycatia-projects\
# first we'll create the project that uses pycatia from pypi
mkdir pycatia-pypi
cd .\pycatia-pypi\
# install and activate the virtualenv
python -m virtualenv env
.\env\Scripts\activate
pip install pycatia
```

Resultant folder structure:

[image]

Each environment (`pycatia-git`, `pycatia-pypi`) should be opened 
independently with their respective virtualenvs activated within your IDE.

ptm-tm, 2023-10-12:
pls note me what  can i do, if i need to see mistake error.

im create venv. and clone my git project in any folder 
than i dev install my module pip install -e "folder with git prj"
and continue develop in git folder without install to side package.
or u can add your git folder in python path, but that case u will see pylint error but all will work.

ptm-tm, 2023-10-12:
pip install -e "git project folder" dont install your project, only make symlink to folder in pyhton path... every time when you run your script in your enviroment