evereux/pycatia #165: Addtional information for contributors

deloarts, 2023-10-12:
@evereux: Maybe add additional information to the contributing chapter in the readme for people who want to contribute to **pycatia**:

- how to format the code
- how to sort imports
- what line length limit is defined
- *something else?*

--------------------------------------------------

For everyone who uses **VS Code**, add this to your workspace settings to match **pycatia**s formatting:

```json
{
    "[python]": {
        "editor.rulers": [
            120
        ],
        "editor.formatOnSave": true,
        "editor.codeActionsOnSave": {
            "source.organizeImports": true
        },
        "editor.defaultFormatter": "ms-python.black-formatter"
    },
    "pylint.args": [
        "--line-length",
        "120"
    ],
    "black-formatter.args": [
        "--line-length",
        "120"
    ],
    "isort.args": [
        "--profile",
        "black",
        "-sl"
    ],
    "python.testing.pytestArgs": [
        "tests/in"
    ]
}
```

Note: This requires the following extensions:

- https://marketplace.visualstudio.com/items?itemName=ms-python.black-formatter
- https://marketplace.visualstudio.com/items?itemName=ms-python.isort
- https://marketplace.visualstudio.com/items?itemName=ms-python.pylint

evereux, 2023-10-12:
The `CONTRIBUTING.md` file which is linked to in the `README.md` is the better place for this kind of thing. It already covers some of this. Which people aren't likely to read anyway from my experience. 😄 

No VSCode settings though as the information should be applicable regardless of IDE.

deloarts, 2023-10-12:
Yes, the `contributing.md` is the better place for that (and yes, I didn't read it ...).

> No VSCode settings though as the information should be applicable regardless of IDE.

I didn't mean to add the VS Code settings to the contributing file, I just left it here in the issue so others can apply the pycatia-formatting-rules directly to their workspace.

closed 2025-08-29 by evereux