evereux/pycatia #166: How can edit links by Pycatia

hosseintfr, 2023-10-13:
I want to edit design table path using python. I know that I can edit a design table path using "Edit">"links" in Catia. But I want to do this using python code. How can I change the directory of a design table using pycatia?

evereux, 2023-10-13:
I didn't have a clue and had to re-teach myself what design tables were. It's been a few years since I used one. Never in a scripting sense.

```python
from pycatia import catia
from pycatia.knowledge_interfaces.design_table import DesignTable
from pycatia.mec_mod_interfaces.part import Part

caa = catia()
application = caa.application
active_document = application.active_document
part = Part(active_document.part.com_object)
parameters = part.parameters
relations = part.relations

relation_design_table = relations.get_item_by_name('DesignTable.2')
design_table = DesignTable(relation_design_table.com_object)
print(design_table.file_path)
```

There's lots of stuff in pycatia I don't have a clue about. People seem to assume I do. 😄

hosseintfr, 2023-10-13:
For those who read this issue: 
If your part has a design table and you want to change its directory using python, you can use this code:

from pycatia import catia
from pycatia.mec_mod_interfaces.part_document import PartDocument
from pycatia.mec_mod_interfaces.part import Part
from pycatia.knowledge_interfaces.design_table import DesignTable

caa = catia()
documents = caa.documents
documents.open(r"D:\Catia_workshop\Part1.CATPart")
part_doc = PartDocument(caa.active_document.com_object)
part = Part(part_doc.part.com_object)
relations = part.relations
relation_design_table = relations.get_item_by_name('DesignTable.1')
design_table = DesignTable(relation_design_table.com_object)
parameters = part.parameters
part.update()
print(design_table.file_path)
design_table.file_path = "D:\Catia_workshop\DesignTable22.xlsx"
print(design_table.file_path)

evereux, 2023-10-13:
> One more question is that, do you think we can change some length and angles of our part in python without using design table? I mean I want to open a part by python (and not create it by python). Then I want to edit some of its parameters by python. Do you think this is possible or not? (I can create a new issue for this if it is necessary.)

You mean update a parameter that doesn't have a design table linked to it? Yeah sure. Think there are examples that should help with that.

With regards to your example above. Do you need to do a `part.update' after changing the design table link? Or does the part update auto-magically?

hosseintfr, 2023-10-13:
1- Yes. Exactly. I mean that I want to update (change) a parameter that is not linked to a design table. I think this way is a better way for designing automation with Catia using python, in comparison with design table. @evereux , please if you have an exaple code send me. It will be very helpful for me.

2- I just used the part.update() command to be sure that it is updated. Because sometimes it takes some time for Catia to update parts automatically. We can delete this command.

hosseintfr, 2023-10-13:
I started a new issue for my new question entitled: 
Change a part's parameters using pycatia (which part that is not created using python)

evereux, 2023-10-25:
Closing as issue seems to be resolved.

closed 2023-10-25 by evereux