evereux/pycatia #167: Change a part's parameters using pycatia (which part that is not created using python)

hosseintfr, 2023-10-13:
I want to update (change) a parameter that is not linked to a design table. I think this way is a better way for designing automation with Catia using python, in comparison with using design table. Please if you have an example code send it to me. It will be very helpful for me.

evereux, 2023-10-13:
Take a look at the [examples](https://pycatia.readthedocs.io/en/latest/) specfically the parameters section.

I recommend working through a number of them so you get a better idea of how to use pycatia.

If you can't get it to work you'll need to provide some example code to show what you've tried.

hosseintfr, 2023-10-13:
@evereux : Unfortunately, I could not understand the example_parameters. I don't have any code. But I can send a Catia part and please open it by python and then change one of its parameters by pycatia. 
[Part1.zip](https://github.com/evereux/pycatia/files/12895556/Part1.zip)

hosseintfr, 2023-10-13:
Thanks @evereux . I read the parameters example and wrote my code. However it just print the parameters names and not the values. I want to see the values and change them. My code is this:

from pycatia import catia
from pycatia.mec_mod_interfaces.part import Part
from pycatia.mec_mod_interfaces.part_document import PartDocument

caa = catia()
documents = caa.documents
documents.open(r"D:\Catia_workshop\Part1.CATPart")
part_doc = PartDocument(caa.active_document.com_object)
part = Part(part_doc.part.com_object)
bodies = part.bodies
part_parameters = part.parameters

root_parameter_set      = part_parameters.root_parameter_set
parameter_sets          = root_parameter_set.parameter_sets
sub_parameter_set       = parameter_sets.parameter_sets

for i in sub_parameter_set:
    sub_parms = i.all_parameters
    for j in sub_parms:
        print(j)
        print("Parameter name: {} - Value: {}".format(j.name, j.value))
    print()

part.update()
part_doc.save()
part_doc.close()

hosseintfr, 2023-10-13:
Thanks @evereux . I read the parameters example and wrote my code. However it just print the parameters names and not the values. I want to see the values and change them. My code is this:

from pycatia import catia
from pycatia.mec_mod_interfaces.part import Part
from pycatia.mec_mod_interfaces.part_document import PartDocument

caa = catia()
documents = caa.documents
documents.open(r"D:\Catia_workshop\Part1.CATPart")
part_doc = PartDocument(caa.active_document.com_object)
part = Part(part_doc.part.com_object)
bodies = part.bodies
part_parameters = part.parameters

root_parameter_set = part_parameters.root_parameter_set
parameter_sets = root_parameter_set.parameter_sets
sub_parameter_set = parameter_sets.parameter_sets

for i in part_parameters:
print(i.name)

for i in sub_parameter_set:
sub_parms = i.all_parameters
for j in sub_parms:
print(j)
print("Parameter name: {} - Value: {}".format(j.name, j.value))
print()

part.update()
part_doc.save()
part_doc.close()

evereux, 2023-10-14:
I've added a new example to show how to do this.

Don't forget, recording macros and reading the output is a good way to figure this out. I'd forgotten and it's what I had to do.

The step you're missing in the initialisation of the parameter type (Length, String, Real ....) from where you can change it's value.

hosseintfr, 2023-10-14:
Thanks a lot.

evereux, 2023-10-25:
Closing as issue seems to be resolved.

closed 2023-10-25 by evereux