evereux/pycatia #171: Edit parts' links with Pycatia

hosseintfr, 2023-10-24:
Its obvious that when you copy a project folder (containing some parts and a product) to another path, its product will read its parts from first directory. You can go to "Edit">"Links" and change the parts' paths. My question is: how can we do this (changing parts links/directories) using Pycatia codes?

evereux, 2023-10-24:
Have you tried doing this manually and recording a macro?

I'm not sure fixing it with a script is the best way to go about this anyway? This can usually be resolved by editing your Document Environment / Linked Document Localization -> search orders etc.

hosseintfr, 2023-10-24:
@evereux Yes I tried it by recording a macro but it doesn't have any beneficial VB code for me. You mean that pycatia connot edit links of parts in a product?!

evereux, 2023-10-25:
> Yes I tried it by recording a macro but it doesn't have any beneficial VB code for me.

It could be beneficial for me. It also shows me what you have tried.

> You mean that pycatia connot edit links of parts in a product?!

Not at all. Just trying to get more information.

 Providing the VBA / CATScript would do that and it turns out in this case it certainly does help.

```Sub CATMain()

Dim productDocument1 As ProductDocument
Set productDocument1 = CATIA.ActiveDocument

Dim product1 As Product
Set product1 = productDocument1.Product

Dim products1 As Products
Set products1 = product1.Products

Dim product2 As Product
Set product2 = products1.Item("SubProduct_1.1")

Dim products2 As Products
Set products2 = product2.Products

Dim product3 As Product
Set product3 = products2.Item("Part1.13")

Set Nothing1 = products2.ReplaceComponent(product3, "new_part.CATPart", True)

End Sub
```

Which leads me to believe what you need is [Products.replace_component()](https://pycatia.readthedocs.io/en/latest/api/pycatia/product_structure_interfaces/products.html#pycatia.product_structure_interfaces.products.Products.replace_component).

hosseintfr, 2023-10-25:
Exactly. Thanks a lot @evereux . But my code is not working. I tried this command:
Products.replace_component(self=Products,i_old_component=partb,i_file_path="D:\Catia_workshop\Partc.CATPart",i_multi_instances=True)

My code is:

from pycatia import catia
from pycatia.mec_mod_interfaces.part import Part
from pycatia.mec_mod_interfaces.part_document import PartDocument
from pycatia.product_structure_interfaces.product import Product
from pycatia.product_structure_interfaces.product_document import ProductDocument
from pycatia.product_structure_interfaces.products import Products

caa = catia()
documents = caa.documents
documents.open(r"D:\Catia_workshop\Productab.CATProduct")
product_doc = ProductDocument(caa.active_document.com_object)
product = Product(product_doc.product.com_object)

documents.open(r"D:\Catia_workshop\Partb.CATPart")
part_doc = PartDocument(caa.active_document.com_object)
partb = Part(part_doc.part.com_object)

Products.replace_component(self=Products,i_old_component=partb,i_file_path="D:\Catia_workshop\Partc.CATPart",i_multi_instances=True)

I want to replace Partc with Partb. But I do not know how to use Products.replace_component command.

hosseintfr, 2023-10-25:
Thanks a lot @evereux . It works!

evereux, 2023-10-25:
Excellent. Glad you got it working.

evereux, 2023-11-03:
Closing as issue is now resolved.

closed 2023-11-03 by evereux