evereux/pycatia #175: [BUG] Cannot add FirstGroup to Clash

Raph-xyz, 2023-11-03:
**Describe the bug**
When creating a clash, in CatClashComputationTypeBetweenTwo mode, we can add the SecondGroup, but not the FirstGroup
it comes from .\pycatia\space_analyses_interfaces\clash.py
```
    @first_group.setter
    def first_group(self, value: Group):
        self.clash.FirstGroup = value
```
which should be : 
```
    @first_group.setter
    def first_group(self, group: Group):
        self.clash.FirstGroup = group.com_object
```

I understood this from the second group setter, wich is already correct:
```
    @second_group.setter
    def second_group(self, group: Group):
        self.clash.SecondGroup = group.com_object
```

**To Reproduce**
here is the simplest code I could do to reproduce: 
```
import pycatia
from pycatia.product_structure_interfaces.product_document import ProductDocument
from pycatia.product_structure_interfaces.product import Product
from pycatia.navigator_interfaces.groups import Groups
from pycatia.navigator_interfaces.group import Group
from pycatia.space_analyses_interfaces.clashes import Clashes

caa = pycatia.catia()
caa.documents.add("Product")
product_doc = ProductDocument(caa.active_document.com_object)
top_node = Product(product_doc.product.com_object)

groups: Groups = Groups(top_node.get_technological_object("Groups"))
group1: Group = groups.add()
group2: Group = groups.add()

clashes = Clashes(top_node.get_technological_object("Clashes"))
clash = clashes.add()
clash.computation_type = 3  # CatClashComputationTypeBetweenTwo
print("Clash created, add groups")
clash.first_group = group1 # NOT WORKING
clash.second_group = group2 # WORKING
print("Groups added to clash")

caa.active_document.close()
```

**Expected behavior**
FirstGroup is set

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - CATIA V5-6R2017
 - pycatia version 0.6.3

evereux, 2023-11-03:
Thanks for the bug report. This isn't something I'm familar with but I don't think the problem lies where you suggest? 

```    
@first_group.setter
    def first_group(self, value: Group):
        """
        :param Group value:
        """

        self.clash.FirstGroup = value
```

The parameter `value`  could be named anything. It doesn't matter whether it's named `value` or `group` (I will change that though for consistency with `second_group()`. I don't currently understand what the issue is though. I'll do some more digging...

Do you happen to have an example of an equivalent working VB/CAT script?

Raph-xyz, 2023-11-03:
Yes sorry I renamed it out of habit, but it's the .com_object that matters here :)

evereux, 2023-11-03:
I've got that fixed now in the development branch.

I'll release 0.6.4 shortly.

Thanks again for this. Much appreciated. 👍

evereux, 2023-11-03:
0.6.4 has been released.

If you installed via pip you can do `pip install --upgrade pycatia`.

closed 2023-11-03 by evereux