evereux/pycatia #176: Weight / mass by pycatia

hosseintfr, 2023-11-30:
How can get weight (or mass) of a part by Pycatia? Is there any python code to get mechanical properties?

evereux, 2023-11-30:
Yes. Please take a look through the examples.

hosseintfr, 2023-11-30:
Thanks a lot. I apologize that I did not check the examples. I understand how to get the mass of each part in a product by example product_3 (example__product__003.py):

from pycatia import catia
from pycatia.enumeration.enumeration_types import cat_work_mode_type
from pycatia.product_structure_interfaces.product import Product
from pycatia.product_structure_interfaces.product_document import ProductDocument

caa = catia()
documents = caa.documents
# documents.open(r"tests\cat_files\product_top.CATProduct")
document = ProductDocument(caa.active_document.com_object)
product = Product(document.product.com_object)
product.apply_work_mode(cat_work_mode_type.index("DESIGN_MODE"))

def print_properties(obj: Product):
    print(
        f"{obj.name}: mass: {obj.analyze.mass}, \n"
        f"    volume: {obj.analyze.volume}, \n"
        f"    wet_area: {obj.analyze.wet_area}, \n"
        f"    gravity_center: {obj.analyze.get_gravity_center()}, \n"
        f"    inertia: {obj.analyze.get_inertia()}"
    )

for sub_product in product.products:
    if sub_product.is_catproduct():
        for child_product in sub_product.products:
            if child_product.is_catpart():
                child_product.activate_default_shape()
                print_properties(child_product)

    else:
        sub_product.activate_default_shape()
        print_properties(sub_product)

closed 2023-11-30 by hosseintfr