evereux/pycatia #178: [BUG]

ZackZouJ, 2023-12-09:
**Describe the bug**
I have tried to run script "example__hybrid_shape_factory__006.py" but got the error 
 raise CATIAApplicationException(f'Could not find hybrid_body "i_index"')
pycatia.exception_handling.exceptions.CATIAApplicationException: Could not find hybrid_body "i_index"
the error line in the sample is 
construction_geometry = hybrid_bodies.item("ConstructionGeometry")
could you please give some advice, thank you 

**To Reproduce**
Steps taken to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Win10 22H2
 - CATIA V5 version V5-6R2020
 - pycatia version  pycatia 0.6.4
 - python version 3.9.13

**Additional context**
Add any other context about the problem here.

evereux, 2023-12-09:
You don't appear to have a geometrical set called `ConstructionGeometry`.

From the header of `example__hybrid_shape_factory__006.py`:

```

    Example - Hybrid Shape Factory - 006

    Description:
        How to add a new sphere when a reference axis system isn't required.

    Requirements:
        - A geometrical set named "ConstructionGeometry".
        - A point within the ConstructionGeometry named "Point.1".

```

ZackZouJ, 2023-12-09:
thanks a lot for the reply, i'm a beginer of pycatia, below is the script that used, i have created on geometrical set with this name, but still got the error , i have tried many times but can't solve the problem that meet, could you please have some guidence, my goal it drawing a ball in CATIA with pycatia for the first step, thank you 
import os
import sys

sys.path.insert(0, os.path.abspath("..\\pycatia"))

from pycatia import catia
from pycatia.mec_mod_interfaces.part import Part
from pycatia.mec_mod_interfaces.part_document import PartDocument
from pycatia.scripts.vba import vba_nothing
from pycatia.hybrid_shape_interfaces.hybrid_shape_factory import HybridShapeFactory
import pythoncom

caa = catia()
documents = caa.documents
documents.add("Part")
document = PartDocument(caa.active_document.com_object)
part = Part(document.part.com_object)

hybrid_bodies = part.hybrid_bodies
geom_set = hybrid_bodies.add()
geom_set.name = "Construction_Geometry"

hybrid_shapes = geom_set.hybrid_shapes
centre_point = hybrid_shapes.item("Point.1")
reference_centre_point = part.create_reference_from_object(centre_point)

hybrid_shape_factory = part.hybrid_shape_factory
radius = 100
begin_parallel_angle = -45
end_parallel_angle = 45
begin_meridian_angle = 0
end_meridian_angle = 180
hybrid_shape_sphere = hybrid_shape_factory.add_new_sphere(
    reference_centre_point,
    vba_nothing,
    radius,
    begin_parallel_angle,
    end_parallel_angle,
    begin_meridian_angle,
    end_meridian_angle
)

# set limitation to 0 for a partial sphere, 1 for a complete sphere.
hybrid_shape_sphere.limitation = 0

geom_set.append_hybrid_shape(hybrid_shape_sphere)

part.update()

ZackZouJ, 2023-12-09:
i have checked again and find the real reason is I have created wrong point, now , it's working, thanks a lot, this topic can be closed

closed 2023-12-09 by ZackZouJ

evereux, 2023-12-09:
Great. Glad you got it worked out.