evereux/pycatia #182: [BUG] Error:  when we're trying to create reference from user selection Edge using function : [create_reference_from_b_rep_name]

siddhu2310, 2023-12-29:
**Describe the bug**
I'm working to create a center point using user Selection Edge. Once we selected Shape and Edge, It will be passed to function `create_reference_from_b_rep_name(str,shapeOjb)`, If we pass the string directly to the function, the reference is created, but if passing the string using a string variable. Getting Error.

```An error occurred: (-2147352567, 'Exception occurred.', (0, 'CATIAPart', 'The method CreateReferenceFromBRepName failed', None, 0, -2147024809), None)```

**To Reproduce**
Use the Below code:
```
from pycatia import catia
from pycatia.mec_mod_interfaces.part import Part
from pycatia.mec_mod_interfaces.shapes import Shapes
caa = catia()
activ_doc= caa.active_document
part_doc = Part(activ_doc.part.com_object)

def select_and_get_object(type):
    usel = caa.active_document.selection
    usel.clear()
    filter = (type,)
    caa.message_box(f"Select the {filter[0]}", 64, "CATIA Automation")
    oStatus = usel.select_element2(filter, f"Select the {filter[0]}", True)
    # print(oStatus)
    
    if not oStatus == "Cancel":
        oSelObj = usel.item(1).value.name
        usel.clear()
    else:
        oSelObj = None
        usel.clear()

    return oSelObj

bodyObj = select_and_get_object("Body")

solidObj = part_doc.bodies.get_item(bodyObj)
print(solidObj.name)
shapes = Shapes(solidObj.shapes.com_object)

shapeObj = shapes.get_item_by_index(1)
print(shapeObj.name)

edgeObj = select_and_get_object("Edge")

print(edgeObj)

edgeObj = edgeObj.replace("Selection_", "")
edgeObj = edgeObj[:edgeObj.rfind("));")]
edgeObj = edgeObj + "))"

print(edgeObj)

try:
    # Gettting Error
    # refEdge = part_doc.create_reference_from_b_rep_name(str(edgeObj), shapeObj)

    # Gettting No Error if we pass directly      
    refEdge = part_doc.create_reference_from_b_rep_name("REdge:(Edge:(Face:(Brp:(Solid.3;%19);None:();Cf12:());Face:(Brp:(Solid.3;%3);None:();Cf12:());None:(Limits1:();Limits2:());Cf12:())", shapeObj)
except Exception as e:
    print(f"An error occurred: {e}")

add_point = part_doc.hybrid_shape_factory.add_new_point_center(refEdge)

geo_set = part_doc.hybrid_bodies.item("Geometrical Set.2")
geo_set.append_hybrid_shape(add_point)
part_doc.in_work_object = add_point
part_doc.update()
```

**Expected behavior**
It should work , when trying to pass as  a string variable.

**Desktop (please complete the following information):**
 - OS: Windows 11
 - CATIA V5 version :  V5-6R2020
 - pycatia version : pycatia 0.6.4

Kindly support to resolve the issue.

evereux, 2023-12-29:
I can't reproduce your error. `create_reference_from_b_rep_name()` worked for me with your variable `edgeObj` passed to it.

I can't think why it wouldn't work for you. The only thing I would double check is to see if your manually generated string exactly matches the variable? 

I don't _think_ the CATIA version matters here but I'll see if I can get a friend to test this in R2020. This won't be for a week or two though.

What version of pywin32 are you using? Mine is `pywin32==305`.

Side note: Your script mostly worked for me. It failed for me on `add_new_point_center` the error coming from updating the part as CATIA doesn't like adding a point to a solid edge like this. At least in my version.

siddhu2310, 2023-12-29:
I'm Surprised, the code working now, I did not change anything,  
Thanks for your effort. 
PyCATIA is great for CATIA automation users ,

closed 2023-12-29 by siddhu2310