evereux/pycatia #184: [BUG] AddNewRotate2 function is not available in ShapeFactory

siddhu2310, 2024-01-05:
**Describe the bug**
add_new_rotate_2 function is not available,
I have VBA code, I'm doing it in Python.

`code:`
```
Sub CATMain()

Debug.Print Now()
Dim partDocument1 As partDocument
Set partDocument1 = CATIA.ActiveDocument

Dim part1 As part
Set part1 = partDocument1.part

Dim bodies1 As Bodies
Set bodies1 = part1.Bodies

Dim body1 As Body
Set body1 = bodies1.Item("test")

part1.InWorkObject = body1

Dim shapeFactory1 As ShapeFactory
Set shapeFactory1 = part1.ShapeFactory

Dim hybridBodies1 As HybridBodies
Set hybridBodies1 = part1.HybridBodies

Dim hybridBody1 As HybridBody
Set hybridBody1 = hybridBodies1.Item("Inertia")

Dim hybridShapes1 As HybridShapes
Set hybridShapes1 = hybridBody1.HybridShapes

Dim hybridShapeAxisLine1 As HybridShapeAxisLine
Set hybridShapeAxisLine1 = hybridShapes1.Item("Axis.23")

Dim reference1 As Reference
Set reference1 = part1.CreateReferenceFromObject(hybridShapeAxisLine1)

rotateAngle = 3

Dim rotate1 As Rotate
'this Add new Rotate is not availble
Set rotate1 = shapeFactory1.AddNewRotate2(reference1, rotateAngle)

Dim hybridShapeRotate1 As HybridShapeRotate
Set hybridShapeRotate1 = rotate1.HybridShape

hybridShapeRotate1.RotationType = 0

hybridShapeRotate1.Axis = reference1

part1.InWorkObject = hybridShapeRotate1

part1.Update

End Sub
```

`Python code:`
```
from pycatia import catia
from pycatia.mec_mod_interfaces.part import Part
from pycatia.mec_mod_interfaces.shapes import Shapes
from pycatia.part_interfaces.shape_factory import ShapeFactory

caa = catia()

activ_doc= caa.active_document
part_doc= Part(activ_doc.part.com_object)

def get_geo_set(part_doc:Part,item:str):
    try:
        geo_set = part_doc.hybrid_bodies.get_item_by_name(item)
    except:
        geo_set = part_doc.hybrid_bodies.add()
        geo_set.name = item
        part_doc.update()
    return geo_set

geo_set_force = get_geo_set(part_doc,"Forces")
geo_set_cables = get_geo_set(part_doc,"Cables")
geo_set_inertia= get_geo_set(part_doc,"Inertia")

rotate_body = part_doc.bodies.get_item_by_name("lever")
fix_body = part_doc.bodies.get_item_by_name("fix")
sh_factory = ShapeFactory(part_doc.shape_factory.com_object)

part_doc.in_work_object = rotate_body

axis_rotate = geo_set_inertia.hybrid_shapes.get_item_by_name("Axis.16")
print(axis_rotate)
ref_axis = part_doc.create_reference_from_object(axis_rotate)
rotate = sh_factory.add_new_rotate2(ref_axis,3.5)

part_doc.update()
```

**Desktop (please complete the following information):**
 - OS: [e.g. Windows 11]
 - CATIA V5 version [V5-6R2020]
 - pycatia version : latest

**Additional context**
Kindly support to achieve this.

evereux, 2024-01-05:
Another feature not documented in the CAA V5 basic help. Thanks for reporting.

For my reference from the MS VB Object viewer so I can implement this:

```
Function AddNewRotate2(iAxis As Reference, iAngle As Double) As AnyObject
    Member of PARTITF.ShapeFactory
```

siddhu2310, 2024-01-06:
> Another feature not documented in the CAA V5 basic help. Thanks for reporting.
> 
> For my reference from the MS VB Object viewer so I can implement this:
> 
> ```
> Function AddNewRotate2(iAxis As Reference, iAngle As Double) As AnyObject
>     Member of PARTITF.ShapeFactory
> ```

Please fix it as soon possible, i'm working on the automation that require this function.
Thanks in Advance.

siddhu2310, 2024-01-08:
I have implemented manually in PARTITF.ShapeFactory
`it should be as Rotate `
```
    def add_new_rotate2(self, i_axis: Reference, i_angle: float) -> Rotate:
        """
        .. note::
            #siddharth
        """
        return Rotate(self.shape_factory.AddNewRotate2(i_axis.com_object, i_angle))
```

evereux, 2024-01-10:
I've added this now to the development branch. There will be a new release before the end of the week hopefully.

evereux, 2024-01-13:
0.6.5 now released which includes this missing class method.

closed 2024-01-13 by evereux

siddhu2310, 2024-01-18:
Kindly Update it on PIP install, It is still 0.6.4 only