evereux/pycatia #187: Creating a product and a part in it by Pycatia

hosseintfr, 2024-01-15:
Hi. I want to create a product and then create a part in it and then start to draw sketch in my part. But I have an error.

My code is this:

import os
import sys

sys.path.insert(0, os.path.abspath("..\\pycatia"))
##########################################################

from pycatia import catia
from pycatia.in_interfaces.reference import Reference
from pycatia.mec_mod_interfaces.body import Body
from pycatia.mec_mod_interfaces.part import Part
from pycatia.mec_mod_interfaces.part_document import PartDocument
from pycatia.product_structure_interfaces.product import Product
from pycatia.product_structure_interfaces.product_document import ProductDocument

caa = catia()
documents = caa.documents

documents.add("Product")
product_doc = ProductDocument(caa.active_document.com_object)
product = Product(product_doc.product.com_object)
product.part_number = "Product_1"

Part_1 = product.products.add_new_component("Part", "Part_1")

document = PartDocument(Part_1.com_object)
part = Part(document.part.com_object)

evereux, 2024-01-15:
[Products.add_new_component](https://pycatia.readthedocs.io/en/latest/api/pycatia/product_structure_interfaces/products.html#pycatia.product_structure_interfaces.products.Products.add_new_component) returns a `Product` object not `Part` which is how you're subsequently treating it.

Record a macro doing this manually and it'll be easy to figure out.

closed 2024-01-15 by evereux

hosseintfr, 2024-01-15:
Thanks @evereux . I recorded a macro but I could not figure out how to fix this problem. This is the macro:

Language="VBSCRIPT"

Sub CATMain()

Dim documents1 As Documents
Set documents1 = CATIA.Documents

Dim partDocument1 As Document
Set partDocument1 = documents1.Item("Part_1.CATPart")

Dim part1 As Part
Set part1 = partDocument1.Part

Dim bodies1 As Bodies
Set bodies1 = part1.Bodies

Dim body1 As Body
Set body1 = bodies1.Item("PartBody")

Dim sketches1 As Sketches
Set sketches1 = body1.Sketches

Dim originElements1 As OriginElements
Set originElements1 = part1.OriginElements

Dim reference1 As AnyObject
Set reference1 = originElements1.PlaneXY

Dim sketch1 As Sketch
Set sketch1 = sketches1.Add(reference1)

Dim arrayOfVariantOfDouble1(8)
arrayOfVariantOfDouble1(0) = 0.000000
arrayOfVariantOfDouble1(1) = 0.000000
arrayOfVariantOfDouble1(2) = 0.000000
arrayOfVariantOfDouble1(3) = 1.000000
arrayOfVariantOfDouble1(4) = 0.000000
arrayOfVariantOfDouble1(5) = 0.000000
arrayOfVariantOfDouble1(6) = 0.000000
arrayOfVariantOfDouble1(7) = 1.000000
arrayOfVariantOfDouble1(8) = 0.000000
sketch1.SetAbsoluteAxisData arrayOfVariantOfDouble1

part1.InWorkObject = sketch1

Dim factory2D1 As Factory2D
Set factory2D1 = sketch1.OpenEdition()

End Sub

evereux, 2024-01-15:
Your macro doesn't create a new product, add a new part and work in a sketch within it though.

This is what I got:

```
Sub CATMain()

Dim documents1 As Documents
Set documents1 = CATIA.Documents

Dim productDocument1 As ProductDocument
Set productDocument1 = documents1.Add("Product")

Dim product1 As Product
Set product1 = productDocument1.Product

Dim products1 As Products
Set products1 = product1.Products

Dim product2 As Product
Set product2 = products1.AddNewComponent("Part", "")

Dim partDocument1 As PartDocument
Set partDocument1 = documents1.Item("Part1.CATPart")

Dim part1 As Part
Set part1 = partDocument1.Part

Dim bodies1 As Bodies
Set bodies1 = part1.Bodies

Dim body1 As Body
Set body1 = bodies1.Item("PartBody")

Dim sketches1 As Sketches
Set sketches1 = body1.Sketches

Dim originElements1 As OriginElements
Set originElements1 = part1.OriginElements

Dim reference1 As Reference
Set reference1 = originElements1.PlaneYZ

Dim sketch1 As Sketch
Set sketch1 = sketches1.Add(reference1)

Dim arrayOfVariantOfDouble1(8)
arrayOfVariantOfDouble1(0) = 0#
arrayOfVariantOfDouble1(1) = 0#
arrayOfVariantOfDouble1(2) = 0#
arrayOfVariantOfDouble1(3) = 0#
arrayOfVariantOfDouble1(4) = 1#
arrayOfVariantOfDouble1(5) = 0#
arrayOfVariantOfDouble1(6) = 0#
arrayOfVariantOfDouble1(7) = 0#
arrayOfVariantOfDouble1(8) = 1#
Set sketch1Variant = sketch1
sketch1Variant.SetAbsoluteAxisData arrayOfVariantOfDouble1

part1.InWorkObject = sketch1

Dim factory2D1 As Factory2D
Set factory2D1 = sketch1.OpenEdition()

Dim geometricElements1 As GeometricElements
Set geometricElements1 = sketch1.GeometricElements

Dim axis2D1 As Axis2D
Set axis2D1 = geometricElements1.Item("AbsoluteAxis")

Dim line2D1 As Line2D
Set line2D1 = axis2D1.GetItem("HDirection")

line2D1.ReportName = 1

Dim line2D2 As Line2D
Set line2D2 = axis2D1.GetItem("VDirection")

line2D2.ReportName = 2

Dim point2D1 As Point2D
Set point2D1 = factory2D1.CreatePoint(76.787737, -20.921239)

point2D1.ReportName = 3

point2D1.Construction = False

sketch1.CloseEdition

part1.InWorkObject = body1

part1.Update

End Sub
```

evereux, 2024-01-15:
If you're not working in a pycatia development environment you probably don't need:

```python

import os
import sys

sys.path.insert(0, os.path.abspath("..\pycatia"))
##########################################################
```

This is to fix issues with relative imports such as `from pycatia import catia`.