evereux/pycatia #189: Reference Axis by Pycatia

hosseintfr, 2024-01-20:
Hi. I want to create a plane with an angle about zx plane. But when I want to define the Z Axis of my part, I get an error.
The macro that I do this is:

Language="VBSCRIPT"

Sub CATMain()

Set documents1 = CATIA.Documents

Set partDocument1 = documents1.Item("Winding_HV.CATPart")

Set part1 = partDocument1.Part

Set originElements1 = part1.OriginElements

Set hybridShapePlaneExplicit1 = originElements1.PlaneZX

Set reference1 = part1.CreateReferenceFromObject(hybridShapePlaneExplicit1)

Set bodies1 = part1.Bodies

Set body1 = bodies1.Item("PartBody")

Set hybridShapes1 = body1.HybridShapes

Set hybridShapeLineExplicit1 = hybridShapes1.Item("Z Axis")

Set reference2 = part1.CreateReferenceFromObject(hybridShapeLineExplicit1)

Set hybridShapeFactory1 = part1.HybridShapeFactory

Set hybridShapePlaneAngle1 = hybridShapeFactory1.AddNewPlaneAngle(reference1, reference2, -30.000000, False)

hybridShapePlaneAngle1.ProjectionMode = False

body1.InsertHybridShape hybridShapePlaneAngle1

part1.InWorkObject = hybridShapePlaneAngle1

part1.Update

End Sub

_________________________________________________________________
My python code is: 
...

Winding_HV = documents.item("Winding_HV.CATPart")
document = PartDocument(Winding_HV.com_object)
part_HV = Part(document.part.com_object)

originElements1 = part_HV.origin_elements
hybridShapePlaneExplicit1 = originElements1.plane_zx
reference1 = part_HV.create_reference_from_object(hybridShapePlaneExplicit1)
bodies1 = part_HV.bodies
body1 = bodies1.item("PartBody")
hybridShapes1 = body1.hybrid_shapes
hybridShapeLineExplicit1 = hybridShapes1.item("Z Axis")
reference2 = part_HV.create_reference_from_object(hybridShapeLineExplicit1)
hybridShapeFactory1 = part_HV.hybrid_shape_factory
hybridShapePlaneAngle1 = hybridShapeFactory1.add_new_plane_angle(reference1, reference2, -30.000000, False)
hybridShapePlaneAngle1.projection_mode = False
part_HV.in_work_object = hybridShapePlaneAngle1

part_HV.update()

hosseintfr, 2024-01-20:
My other way for creating that plane is by defining the plane through 2 lines. The macro of this way works well without any error. But the pycatia method does not work. However, python does not have any error. But no plane is created!

Macro code:
Language="VBSCRIPT"

Sub CATMain()

Dim documents1 As Documents
Set documents1 = CATIA.Documents

Dim partDocument1 As Document
Set partDocument1 = documents1.Item("Winding_HV.CATPart")

Dim part1 As Part
Set part1 = partDocument1.Part

Dim hybridShapeFactory1 As Factory
Set hybridShapeFactory1 = part1.HybridShapeFactory

Dim bodies1 As Bodies
Set bodies1 = part1.Bodies

Dim body1 As Body
Set body1 = bodies1.Item("PartBody")

Dim sketches1 As Sketches
Set sketches1 = body1.Sketches

Dim sketch1 As Sketch
Set sketch1 = sketches1.Item("sketch_line_b")

Dim reference1 As Reference
Set reference1 = part1.CreateReferenceFromObject(sketch1)

Dim sketch2 As Sketch
Set sketch2 = sketches1.Item("sketch_line_a")

Dim reference2 As Reference
Set reference2 = part1.CreateReferenceFromObject(sketch2)

Dim hybridShapePlane2Lines1 As HybridShapePlane2Lines
Set hybridShapePlane2Lines1 = hybridShapeFactory1.AddNewPlane2Lines(reference1, reference2)

body1.InsertHybridShape hybridShapePlane2Lines1

part1.InWorkObject = hybridShapePlane2Lines1

part1.Update

End Sub
-------
Pycatia code:

Winding_HV = documents.item("Winding_HV.CATPart")
document = PartDocument(Winding_HV.com_object)
part_HV = Part(document.part.com_object)
hybridShapeFactory = part_HV.hybrid_shape_factory
bodies = part_HV.bodies
body = bodies.get_item_by_name("PartBody")
sketches1 = body.sketches

sketch_line_a = sketches1.item("sketch_line_a")
sketch_line_b = sketches1.item("sketch_line_b")
reference1 = part_HV.create_reference_from_object(sketch_line_a)
reference2 = part_HV.create_reference_from_object(sketch_line_b)
Plane2Lines = hybridShapeFactory.add_new_plane2_lines(reference1, reference2)

part_HV.in_work_object = Plane2Lines
part_HV.update()

evereux, 2024-01-20:
In your second example you've omitted the equivalent of `body1.InsertHybridShape hybridShapePlane2Lines1`.

In your first example the same thing is missing but you say you're getting an error. What is the error you're getting? How do I need to set-up my CATIA part to run your code?

hosseintfr, 2024-01-20:
There is no command equivalent with body1.InsertHybridShape in Pycatia. Where is inserhybridshape command?!

In my first code my error is this: (it seems that the Z axis is not defined in Catia)

[2024-01-20 16:36:43,944] INFO in documents: Creating a new "Product".
Traceback (most recent call last):
  File "C:\Users\u6790\PycharmProjects\pythonProject\Pycatia_Winding.py", line 798, in <module>
    hybridShapeLineExplicit1 = hybridShapes1.item("Z Axis")
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\u6790\PycharmProjects\pythonProject\venv\Lib\site-packages\pycatia\mec_mod_interfaces\hybrid_shapes.py", line 103, in item
    return HybridShape(self.hybrid_shapes.Item(i_index))
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<COMObject <unknown>>", line 2, in Item
pywintypes.com_error: (-2147352567, 'Exception occurred.', (0, 'CATIAHybridShapes', 'The method Item failed', None, 0, -2147352573), None)

hosseintfr, 2024-01-20:
Can you please create a plane by pycatia that is equivalent of one of three main planes with an angle about it? 
I can just create a plane that is offset of one of three main planes. The other ways for creating plane is not working. Can you please write an example that create a plane other than offset method? Like angle with a plane, or crossing three points or by crossing two lines?

evereux, 2024-01-20:
> There is no command equivalent with body1.InsertHybridShape in Pycatia. Where is inserhybridshape command?!

https://pycatia.readthedocs.io/en/latest/api/pycatia/mec_mod_interfaces/body.html#pycatia.mec_mod_interfaces.body.Body.insert_hybrid_shape !?

https://github.com/evereux/pycatia/blob/7263c27b248e3d2fd107dfd395943625a5410315/pycatia/mec_mod_interfaces/body.py#L200 !?

You need to provide me information so I may reproduce your problem then point you in the right direction. I don't what my part needs in order reproduce your error.

evereux, 2024-01-20:
> Can you please create a plane by pycatia that is equivalent of one of three main planes with an angle about it?
I can just create a plane that is offset of one of three main planes. The other ways for creating plane is not working. Can you please write an example that create a plane other than offset method? Like angle with a plane, or crossing three points or by crossing two lines?

Maybe in the future. I don't have the time right now. Maybe someone else can.

hosseintfr, 2024-01-20:
Thanks a lot. I found my problem. I did not import HybridShapes in the first of my code. 
I added this:
from pycatia.mec_mod_interfaces.body import Body, HybridShape, HybridBodies, Shapes

But yet I don't know how to define the 3 major axis (x, y, and z axis) to pycatia.

evereux, 2024-01-29:
Here is a working example.

requires:
* CATPart open
* Geometrical set named "Geometrical Set.1"
* Inside "Geometrical Set.1" a line named "Line.1" which is drawn normal to the XY plane.

```python
from pycatia import catia
from pycatia.mec_mod_interfaces.part import Part

caa = catia()
application = caa.application
part_document = application.active_document
part = Part(part_document.part.com_object)

bodies = part.bodies
hybrid_bodies = part.hybrid_bodies
hsf = part.hybrid_shape_factory

geom_set = hybrid_bodies.get_item_by_name("Geometrical Set.1")
hybrid_shapes = geom_set.hybrid_shapes
line = hybrid_shapes.get_item("Line.1")

origin_elements = part.origin_elements
plane_zx = origin_elements.plane_zx

ref_line = part.create_reference_from_object(line)
ref_plane_zx = part.create_reference_from_object(plane_zx)

new_plane = hsf.add_new_plane_angle(ref_plane_zx, ref_line, 45, True)
geom_set.append_hybrid_shape(new_plane)

part.update()
```

closed 2024-01-29 by evereux