evereux/pycatia #192: Question about using iObjects from pycatia

CesarAero, 2024-02-26:
Dear pycatia community,

I am facing a difficulty with pycatia and I couldn't find the right place to ask it. I hope that you could provide me some support as I couldn't find the right answer in the pycatia documentation, the CATIA macro recorder or asking ChatGPT...

My problem is linked with the usage of iObjects for hybrid shape operations (Generative Shape Design workbench tools). It is quite simple to use the GUI to project a sketch or to select a plane and make hybrid operations but accessing to the right elements via pycatia is quite challenging. I have created a reproducer of my problem where a simple tasks needs to be achieved: to project a sketch into another plane. Here you can download the pycatia code that I've coded to show my problem (I changed the extension to .txt because I couldn't send .py codes)

[problem_reproducer.txt](https://github.com/evereux/pycatia/files/14406686/problem_reproducer.txt)

I am facing 2 main problems in this reproducer :

1. I cannot use the "plane_XY = part.OriginElements.PlaneXY" variable to create an offset plane using "plane_proj = hsf.add_new_plane_offset(i_plane, i_offset, i_orientation)" because the plane_XY  belongs to "PartBody" and it seems that plane_XY isn't an "i_plane". I couldn't eve use the selection function. Solution: I had to duplicate the plane as a hybrid body.
2. I cannot project the curve in sketch_1 in the new plane using the function "hsf.add_new_project(i_element, i_support)" once again because sketch_1 isn't a iObject. I also tried addressing directly the curve that I want to project by using "sketch_curve = sketch_1.GeometricElements.item("circle_to_project")" but this doesn't make the sketch curve a i_curve.

I am aware that working with sketches and hybrid bodies is challending even in CATIA VB due to the differences in the used workbenches. Which would be the right way to address to the part bodies and use them in hybrid operations? Or should I convert them always into iObjects?

Thank you in advance!

evereux, 2024-02-26:
You need to read the source code / documentation to see what objects a class method requires.

For example [add_new_project](https://pycatia.readthedocs.io/en/latest/api/pycatia/hybrid_shape_interfaces/hybrid_shape_factory.html#pycatia.hybrid_shape_interfaces.hybrid_shape_factory.HybridShapeFactory.add_new_project) requires a reference object, not a sketch object.

Take a look at the examples for creating references as it's a common requirement. This is not a pycatia thing but a CATIA COM interface thing.

CesarAero, 2024-02-27:
Thank you evereux for your message,

I managed to fix the problem of the reference to the XY plane by using the [Reference class](https://pycatia.readthedocs.io/en/latest/api/pycatia/in_interfaces/references.html) and I don’t need anymore to duplicate the plane.

However the problem with projecting the sketch persists. I have also tried to pass the curve inside the sketch to the `add_new_project ` function through the interface functions edge and shape from [pycatia.mec_mod_interfaces]( https://pycatia.readthedocs.io/en/latest/api/index_mec_mod_interfaces.html) as:
`from pycatia.mec_mod_interfaces.shape import Shape`
`sketch_curve = sketch_1.GeometricElements.item("circle_to_project")`
`proj_sketch = hsf.add_new_project(Shape(sketch_curve), plane_proj)`
`proj_sketch.name = "proj_sketch"`
`aux_bodies.append_hybrid_shape(proj_sketch)`

But this operation is not possible:
[image]

I have checked some forums about programing in CATIA and this operation in VB is simply:
`Set reference1 = part.CreateReferenceFromObject(sketch_1)`
`Set reference2 = part.CreateReferenceFromObject(plane_proj)`
`Set hybridShapeProject = hybridShapeFactory.AddNewProject(reference1, reference2)`

There is no need to convert the sketches because I guess that the VB interface takes all this into account.

I couldn't find helpful information about how to use a sketch in hybrid operations using pycatia. Could you give me some advice about how to face this issue? I could in exchange prepare an example script for this specific problem and give it to you so you can post is as example for this particular operation. Then future users can go directly and learn from this script.

evereux, 2024-02-27:
Glad you got it fixed.  👍

evereux, 2024-03-17:
Closing as issue seems to be now resolved.

closed 2024-03-17 by evereux