evereux/pycatia #193: Problem with create_reference_from_name

CesarAero, 2024-02-29:
Dear support,

I need to select a surface that was created **after using a pocket operation** in order to do a projection in a sketch. I have used as inspiration the pycatia example [CREATE LINES NORMAL TO SURFACE](https://github.com/evereux/pycatia/blob/master/user_scripts/create_lines_normal_to_surface.py) to be able to manualy select a surface and get its reference as it follows:

`selection.clear()`
`caa.message_box('Select the cut surface from the blade.', buttons=1, title='Make your selection')`
`selection.select_element2(('BiDim',), 'Please select Surface.', True)`
`surf_blade = selection.item2(1).value`
`selection.clear()`

For this case `surf_blade.name = "Selection_RSur:(Face:(Brp:(Pocket.1;0:(Brp:(Sketch.1;2)));None:();Cf12:());Pad.1_ResultOUT;Z0;G8782)"` and it works perfectly when I use the variable `surf_blade` to project the surface into a sketch as you can see:

[image]

The next step is to skip this manual input by using the function `create_reference_from_name`. I have set a variable with the value `surf_blade = part.create_reference_from_name("Face:(Brp:(Pocket.1;0:(Brp:(Sketch.1;2)));None:();Cf12:());Pad.1_ResultOUT;Z0;G8782")` based on the output from the manual approach (I remove the `Selection_RSur()` part, I was having errors otherwise). However I have the following incorrect projection of a non-existing geometry with this new approach:

[image]

I am using `selection.item2(1).value` and `part.create_reference_from_name()` on the same reference name and this leads to totally different results. Any idea of why this is happening?

evereux, 2024-02-29:
> I am using selection.item2(1).value and part.create_reference_from_name() on the same reference name and this leads to totally different results. Any idea of why this is happening?

Never tried to do this and I'm not sure this approach (automatically selecting solid based features) will be reliable? You would need to see if that could be done in VBA and work back from that.

How about attacking this differently: What is it that's limiting the base of the pocket (looks like an unconventional pocket)? If it's a plane / surface you could use that to intersect the solid then create a reference from that.

Other than that I can't help you here sorry. Maybe someone else can.

CesarAero, 2024-02-29:
Ok I found the solution: **change the pocket limit mode**. I was using the limit mode "Dimension" with a really high value to make sure that the full geometry was being cut (so pycatia was selecting the super large prism face). I changed it to "Up to last" mode and now it selects the correct face:

`pocket1 = shpfac.add_new_pocket(sketch_1, R)`
`pocket1.first_limit.limit_mode = 2 # Up to last`
`pocket1.is_symmetric = True`

Thanks for your help!

evereux, 2024-03-17:
Closing as it seems the issue has now been resolved.

closed 2024-03-17 by evereux