evereux/pycatia #199: Create a 2D spline inside a sketch

CesarAero, 2024-03-13:
Dear support,

I am trying to create a spline inside a sketch by using [create_spline](https://pycatia.readthedocs.io/en/latest/api/pycatia/sketcher_interfaces/factory_2D.html#pycatia.sketcher_interfaces.factory_2D.Factory2D.create_spline) and [create_control_point](https://pycatia.readthedocs.io/en/latest/api/pycatia/sketcher_interfaces/factory_2D.html#pycatia.sketcher_interfaces.factory_2D.Factory2D.create_control_point) with the following code:

`# Create sketch in plane_ZX`
`sketch_3 = sketches.add(plane_ZX)`
`sketch_3.name = "spline"`
`ske2D_3 = sketch_3.open_edition()`

`# Spline coordinates` 
`L=100`
`ctrl_pt_1 = ske2D_3.create_control_point(-L, L/2)`
`ctrl_pt_2 = ske2D_3.create_control_point(0, L/4)`
`ctrl_pt_3 = ske2D_3.create_control_point(L/2, L)`

`spline_tuple = (*ctrl_pt_1,*ctrl_pt_2,*ctrl_pt_3)` **ERROR HERE!**

`ske2D_3.create_spline(spline_tuple)` 

`# sketch close edition `
`sketch_3.close_edition()`

`# Update the document`
`document.part.update()`

However the `spline_tuple` that is fed into the `create_spline` command is not able to contain `sketcher_interfaces.control_point_2D.ControlPoint2D` type objects.

How could I succesfully pass the points into the spline?

Thank you in advance!

evereux, 2024-03-13:
Looks like a bug. 

The create_spline class method needs to be rewritten using a VB method. Not sure when I'll get a chance to do that right now.

mokrueger, 2024-03-13:
What exactly are you trying to do at the `spline_tuple = (*ctrl_pt_1,*ctrl_pt_2,*ctrl_pt_3) `part?

CesarAero, 2024-03-13:
I was trying to create a pytjon tuple since an `i_tuple` is required as input for the function [create_spline(i_poles: tuple)](https://pycatia.readthedocs.io/en/latest/api/pycatia/sketcher_interfaces/factory_2D.html#pycatia.sketcher_interfaces.factory_2D.Factory2D.create_spline)

mokrueger, 2024-03-13:
Just do `spline_tuple = (ctrl_pt_1.com_object, ctrl_pt_2.com_object, ctrl_pt_3.com_object)` that should work for now :)

evereux, 2024-03-13:
Nice. I hadn't looked into this much.

Thanks for this. Great stuff.

mokrueger, 2024-03-13:
You're welcome. You could also take a look at my other 2 PRs if you got the time. :)

evereux, 2024-03-13:
Yeah I saw those, thanks again.

I'm hoping to get these all merged by the end of the week.

mokrueger, 2024-03-13:
@CesarAero after the changes are merged in version 0.6.7, it will work with `spline_tuple = (ctrl_pt_1, ctrl_pt_2, ctrl_pt_3)`. Just so you won't be surprised in the future.

evereux, 2024-03-17:
Closing as improvements made by @mokrueger are now in release 0.6.7.

closed 2024-03-17 by evereux