evereux/pycatia #202: [FEATURE REQUEST] Implementation of add_new_translate operation

CesarAero, 2024-03-14:
Dear support,

I have noticed that the `add_new_translate` body operation is missing in [shape_factory](https://pycatia.readthedocs.io/en/latest/api/pycatia/part_interfaces/shape_factory.html). This body operation simply translates the part in a desired direction for an specified distance. Would it be possible to include this operation for future versions of pycatia?

Kind regards,

César.

evereux, 2024-03-14:
Should that be `add_new_translate2`? 

It looks like this is another undocumented feature (no mention of it in chm help file) so wasn't implemented.

For reference here are the details from the VB Object browser (should someone wish to add before I do):

```
Function AddNewTranslate2(iDistance As Double) As AnyObject
    Member of PARTITF.ShapeFactory

```

CesarAero, 2024-03-14:
It seems that `ShapeFactory` object has no attributes `AddNewTranslate`, `AddNewTranslate2` or `add_new_translate2`. I hope that it isn't hard to implement :)

evereux, 2024-03-14:
It should be straightforward to do. 😊

evereux, 2024-03-17:
Turns out it wasn't so straight forward! I've added the feature in the development branch.

I had a hard time figuring out why I couldn't get it to work in a way I expected and I'm still confused why I have to write this example this way:

```python 

from pycatia import catia
from pycatia.hybrid_shape_interfaces.hybrid_shape_translate import HybridShapeTranslate
from pycatia.in_interfaces.reference import Reference
from pycatia.mec_mod_interfaces.hybrid_shape import HybridShape
from pycatia.mec_mod_interfaces.part_document import PartDocument

caa = catia()
part_document = PartDocument(caa.active_document.com_object)
part = part_document.part
bodies = part.bodies
part_body = bodies.item('PartBody')
part.in_work_object = part_body
hybrid_shape_factory = part.hybrid_shape_factory

origin_elements = part.origin_elements
xy_plane = origin_elements.plane_xy
ref_xy_plane = part.create_reference_from_object(xy_plane)
direction = hybrid_shape_factory.add_new_direction(ref_xy_plane)

shape_factory = part.shape_factory
translate = shape_factory.add_new_translate2(100)
hybrid_shape_translate = HybridShapeTranslate(translate.com_object.HybridShape)

hybrid_shape_translate.vector_type = 0
hybrid_shape_translate.direction = direction

part.update()
```

The quirk being this line 

    hybrid_shape_translate = HybridShapeTranslate(translate.com_object.HybridShape)

I couldn't successfully define the `hybrid_shape_translate.direction` without doing this unconventional call to the `translate.com_object.HybridShape`.

mokrueger, 2024-03-17:
add_new_translate2 returns a Translate object. So I am guessing its intended to access the HybridShapeTranslate like that. Do you know how the Translate class is used in another context? Or is it only used in the context of using its .HybridShape method?

evereux, 2024-03-17:
> add_new_translate2 returns a Translate object

Where did you get that information? All I've got is what I posted above.

I've seen the [Translate](https://pycatia.readthedocs.io/en/latest/api/pycatia/part_interfaces/translate.html) object and I did try initializing that using the AnyObject returned from `add_new_translate2` without any success.

I'll do some more experimenting.

>  So I am guessing its intended to access the HybridShapeTranslate like that. Do you know how the Translate class is used in another context? Or is it only used in the context of using its .HybridShape method?

Not a clue how the Translate class is used in any other context.

evereux, 2024-03-17:
> > add_new_translate2 returns a Translate object

> Where did you get that information? All I've got is what I posted above.

I can see from the generated macro that this is produced:

```
Dim translate1 As Translate
Set translate1 = shapeFactory1.AddNewTranslate2(50#)
```

Which I've tried to reproduce.

evereux, 2024-03-17:
I've updated `add_new_translate()` to now return a Translate object but I still need to what I did above.

This is what I expected to be able to do:

```
hybrid_shape_translate = HybridShapeTranslate(translate.hybrid_shape)
```

You can see there's an issue by trying to `print(hybrid_shape_translate)` with this as that fails too.

evereux, 2024-03-17:
I'm currently out of ideas for this so I'm going to release 0.6.7 today.

I'll keep this issue open in the hope that we can somehow improve `add_new_translate2()` so it's usage makes more sense.

CesarAero, 2024-03-30:
Hi again! Thanks for updating the pycatia version, now I can use the `add_new_translate2` function. However I have been testing it and I cannot pass the arguments correctly to make it work. I can successfully create the part operation in CATIA but it doesn't seem to work even if I complete this operation manually:

[image]

I am using the following code:

`# Set PartBody as a working object`
`part.in_work_object = partbody`

`# Translate direction`
`trans_dir = hsf.add_new_direction_by_coord(1, 1, 0)`

`# Add translate`
`translate1 = shpfac.add_new_translate2(20)`
`translate1.vector_type = 0 # "Direction, distance" translation type`
`translate1.direction = trans_dir`

`# Update the document`
`document.part.update()`

How could I apply this operation correctly? Thank you in advance!

evereux, 2024-04-02:
Did you try recording a macro to see how it's implemented there? It's pretty much always going to be my first question. It's a very helpful tool to help point us in the right direction.

I think not since `translate1.vector_type = 0 # "Direction, distance" translation type` doesn't seem right to me. See the note in [HybridShapeTranslate.vector_type](https://pycatia.readthedocs.io/en/latest/api/pycatia/hybrid_shape_interfaces/hybrid_shape_translate.html#pycatia.hybrid_shape_interfaces.hybrid_shape_translate.HybridShapeTranslate.vector_type).

CesarAero, 2024-04-02:
I tried to record the operation in the CATIA macro recorder and the code was something like this:

`translate1 = shpfac.AddNewTranslate2(20)`
`translate1.VectorType = 0`
`translate1.Direction = trans_dir`

Which is the same as I tried to do but with the corresponding pycatia syntax. It seems to fail when passing the hybrid direction that I created using `trans_dir = hsf.add_new_direction_by_coord(1, 1, 0)` in the following code

`translate1 = shpfac.add_new_translate2(20)`
`translate1.vector_type = 0`
`translate1.direction = trans_dir`

`translate1.vector_type = 0` works for me since 0 is the index for "Direction + distance". I just put a # to add a comment for myself.

evereux, 2024-04-02:
That vector type is definitely wrong if you're trying to translate by co-ordinates. See the documentation I linked to (open the Note).

```
0= Direction + distance
1= point + points
2= coordinates
3= Unknown type
```

Can you post the output of the macro you created please?

evereux, 2024-04-02:
> translate1.vector_type = 0 works for me since 0 is the index for "Direction + distance". I just put a # to add a comment for

But that's not what you want.

evereux, 2024-04-22:
@CesarAero 

Can you clarify here what it is you actually want to do? Reading over these posts again I'm now not sure if you want to translate in a specific direction or translate by co-ordinates? You seem to be mixing two concepts up here?

Anyway, if you can post the full output of a macro recording what you're trying to do that would be helpful.

evereux, 2024-04-22:
Just wanted to add an updated example which is now technically correct:

```python

from pycatia import catia
from pycatia.hybrid_shape_interfaces.hybrid_shape_translate import HybridShapeTranslate
from pycatia.mec_mod_interfaces.part_document import PartDocument

caa = catia()
part_document = PartDocument(caa.active_document.com_object)
part = part_document.part
bodies = part.bodies
part_body = bodies.item('PartBody')
part.in_work_object = part_body
hybrid_shape_factory = part.hybrid_shape_factory

origin_elements = part.origin_elements
xy_plane = origin_elements.plane_xy
ref_xy_plane = part.create_reference_from_object(xy_plane)
direction = hybrid_shape_factory.add_new_direction(ref_xy_plane)

shape_factory = part.shape_factory
translate = shape_factory.add_new_translate2(100)

hybrid_shape_translate = HybridShapeTranslate(translate.hybrid_shape.com_object)

hybrid_shape_translate.vector_type = 0
hybrid_shape_translate.direction = direction

part.update()
```

What I've done here is change `hybrid_shape_translate = HybridShapeTranslate(translate.com_object.HybridShape)` to `hybrid_shape_translate = HybridShapeTranslate(translate.hybrid_shape.com_object)` which makes more sense in my head considering my reference macro. A silly oversight by me.