evereux/pycatia #203: [BUG] set_limitation not working for setting the circle limitations

CesarAero, 2024-03-14:
Dear support,

I believe that I found a bug in the code regarding the `set_limitation` funtion to set the circle limitation type when defining a circle with the hybrid shape factory. First I define the circle as it follows:

`circle_3p = hsf.add_new_circle3_points(p_x, p_z, p1) `
`circle_3p.name = "circle_3p"`
`circle_3p.set_limitation = 2`
`construction_elements.append_hybrid_shape(circle_3p)`

`# Update the document`
`document.part.update()`

But here you can see how I retrieve the limitation value using `get_limitation()` but it doesn't change after setting a different value with `.set_limitation`:

[image]

mokrueger, 2024-03-15:
set_limitation is a function. Does it work with .set_limitation(1)?

CesarAero, 2024-03-15:
You are right, it works now. Sorry for this, I thought that not having an error and not doing anything would mean that there is a bug...

closed 2024-03-16 by evereux