evereux/pycatia #204: Question about create_reference_from_b_rep_name function

CesarAero, 2024-03-15:
Dear support,

I am practicing the use of surface operations and elements selection in the generative shape design workbench. I came accross with a problem that I cannot figure out the solution for it. I will try to explain my situation with a problem reproducer:

[reproduced_BorderREdge.txt](https://github.com/evereux/pycatia/files/14618210/reproduced_BorderREdge.txt)

The previous code creates a `fill` operation in a cylinder with 3 approaches from which the third one doesn't work. The "challenge" here is to pass the **border edge (BEdge)** reference to the [add_new_fill](https://pycatia.readthedocs.io/en/latest/api/pycatia/hybrid_shape_interfaces/hybrid_shape_factory.html#pycatia.hybrid_shape_interfaces.hybrid_shape_factory.HybridShapeFactory.add_new_fill) operation. **In the 1st approach**, I am using the `CATIA Macro recorder` name to reference de selected edge with the [create_reference_from_b_rep_name](https://pycatia.readthedocs.io/en/latest/api/pycatia/mec_mod_interfaces/part.html#pycatia.mec_mod_interfaces.part.Part.create_reference_from_b_rep_name) function and this does the job. **In the 2nd approach**, I am selecting manually the edge and this approach works too. **However the problem comes in the third approach,** were I tried to use the output name from the selection function [select_element2](https://pycatia.readthedocs.io/en/latest/api/pycatia/in_interfaces/selection.html#pycatia.in_interfaces.selection.Selection.select_element2) into [create_reference_from_b_rep_name](https://pycatia.readthedocs.io/en/latest/api/pycatia/mec_mod_interfaces/part.html#pycatia.mec_mod_interfaces.part.Part.create_reference_from_b_rep_name) and it seems to select the edge and pass it to the fill operation but it causes an error in the fill generation as you can observe in the image below:

[image]

How can I correctly pass the reference of this border edge without using the CATIA macro recorder tool to get the reference name and work entirely in pycatia?

evereux, 2024-03-16:
this works for me:

```python

from pycatia import catia
from pycatia.mec_mod_interfaces.part_document import PartDocument

caa = catia()
active_document = PartDocument(caa.active_document.com_object)
part = active_document.part
hsf = part.hybrid_shape_factory
hbs = part.hybrid_bodies
construction_elements= hbs.item("construction_elements")
hs = construction_elements.hybrid_shapes
cylinder = hs.item("Cylinder.1")
reference_edge = part.create_reference_from_b_rep_name("BorderREdge:(BEdge:(Brp:(GSMCylinder.1;2);None:(Limits1:();Limits2:();-1);Cf11:());WithPermanentBody;WithoutBuildError;WithSelectingFeatureSupport;MFBRepVersion_CXR15)", cylinder)

new_fill = hsf.add_new_fill()
new_fill.add_bound(reference_edge)
new_fill.continuity = 0
construction_elements.append_hybrid_shape(new_fill)
part.update()
```

CesarAero, 2024-03-21:
That way works for me too. You used the CATIA macro recorder in VB language to get this reference right? This referencing contains expressions like "WithPermanentBody;WithoutBuildError;WithSelectingFeatureSupport;MFBRepVersion_CXR##". I could work with it but I was just curious of why the output name from the [select_element2](https://pycatia.readthedocs.io/en/latest/api/pycatia/in_interfaces/selection.html#pycatia.in_interfaces.selection.Selection.select_element2) leads to errors sometimes when using this referencing by name :)

evereux, 2024-03-26:
Not a clue. I probably wouldn't do it like this but extract the edges and close them?

Anyways, issue resolved.

closed 2024-03-26 by evereux