evereux/pycatia #206: Get ABS coordinates in Product using get_points_on_axis()

nlhanlha, 2024-04-23:
There's any method to get abs coordinates in product with using get_points_on_axis()?

When I use get_points_on_axis() in product, I can get coordinate about part or child product

i'm trying to solve this problem with many ways, like get_components, np.dot

but if I work on complex product, It isnt easy to handle

Do you have any method to solve it?

Thank you

evereux, 2024-04-23:
I'm not sure what it is you're exactly trying to measure but I think it's the position of the Part in a Product? If so, this should be what you're looking for:

```python

from pycatia import catia
from pycatia.in_interfaces.reference import Reference
from pycatia.product_structure_interfaces.product import Product
from pycatia.product_structure_interfaces.product_document import ProductDocument

caa = catia()
caa.active_document.selection.clear()
product_document = ProductDocument(caa.active_document.com_object)
spa_workbench = product_document.spa_workbench()
product = product_document.product

caa.message_box("Select Surface, you want create the axis", title="Title")

selection = product_document.selection
object_filter = ("Product",)
status = selection.select_element2(object_filter, "Select Surface,  you want create the axis", False)

if status == "Normal":

    selected_element = selection.item(1)
    value = selected_element.value
    product = Product(value.com_object)
    print(product.position.get_components())

else:

    caa.message_box("Please Select/Create GAE Point part before proceeding")

```

nlhanlha, 2024-04-23:
Thank you for your comment.

Well, I'm trying to measure an axis, center of selected surface, in product.

First, If i select a surface in product, i can get an information about points on center axis of surface with using measurable.get_points_on_axis()

However, these points' coords are dependent on child product or part.

So I can't use these coords in whole product workspace(document?).

I tried to move the coordinates of the points using the components of the product which the points are dependent
(using product.position.get_components() and numpy)

But, it also has a problem.

If a point is dependent on the product over several steps, it can not use in this case.

So, I want to know if there any method to get information about points' absolute coordinates on center axis of surface in product

thank you.

evereux, 2024-04-25:
I'm still a bit confused at to what it is you're asking due to the terminology / language barrier I suspect.

> but if I work on complex product, It isnt easy to handle

What do you have that works in a simple product and not a complex product? 

What do you mean by complex product?

> Well, I'm trying to measure an axis, center of selected surface, in product.

I genuinely don't understand what this means. Do you want the center of the selected surface (center of gravity) using the parent product co-ordinate system?

Let's try and tackle this first. The rest of the information is just adding to my confusion.

nlhanlha, 2024-04-30:
hmm.. sorry for my misunderstanding

Until yesterday, I've been mistaken about "get_component" function

there's no problem with any code thank you.

evereux, 2024-04-30:
I'm closing the issue as it seems to be resolved.

closed 2024-04-30 by evereux