evereux/pycatia #207: [FEATURE REQUEST]

ozserhatt, 2024-04-28:
Hello dear Manager !

I can not ( add or append ) to my PartBody a new split. I can not find them in shape.py

for example : 

**if I append to my geometrical set a new split , I can find it ( mec_mod_interfaces --> hybrid_shape --> append_hybrid_shape )

**if I append to my PartBody a new split, I can not find it ( mec_mod_interfaces --> shape --> ??? )

We have only ( __init__ and __repr__ )

Where I can find it or it is a problem with module ?

by the way thank you for your efforts 

Sincerely

evereux, 2024-04-28:
Have a look through the examples or user_scripts. I'm sure I've used a split in one of them. If not, I can take a look tomorrow. 

Have you tried recording a macro doing this manually? That normally gives us many clues on what to do.

ozserhatt, 2024-04-28:
There is a split example in examples it is right but it is about for geometrical set in mechanical design ( mec_mod_interfaces )
If you are append in your body in part design, I can't find for add my split.

Actually I was look from Object Browser in Catia for VB. It also doesn't have any functions append or add. Maybe I can't use this feature. 

"""

    Example - Hybrid Shape Factory - 005

    Description:
        GSD: Split a surface using a plane.

    Requirements:
        - A geometrical set named "ConstructionGeometry".
        - A surface within the geometrical set called "Surface.1" that can be split by the origin ZX plane.

"""
.............

hb_construction_geometry.append_hybrid_shape(hs_split)

"""

ozserhatt, 2024-04-28:
I am sorry dear brother, It doesn't have a function in shape.py already.  
The class shape_factory has the "add" keyword in its function name to add in Catia.

The problem solved. Thank you. You can close this issue if you want. 

Warm regards to you :)

evereux, 2024-04-30:
I'm closing the issue as it seems to be resolved.

closed 2024-04-30 by evereux