evereux/pycatia #208: Help For Pycatia Developer

ozserhatt, 2024-04-30:
Can we ask our own needs or questions here while developing Pycatia? We can ask our code usage help or questions while creating applications here.

evereux, 2024-05-01:
> Can we ask our own needs or questions here while developing Pycatia? We can ask our code usage help or questions while creating applications here.

[https://github.com/evereux/pycatia/issues](Raising Issues / Asking Questions)

Similar questions:

https://github.com/evereux/pycatia/issues/187

https://github.com/evereux/pycatia/issues/51

Also, take a look at https://github.com/evereux/pycatia/blob/master/pycatia/product_structure_interfaces/products.py

And finally, record a macro doing what it is you want to achieve. The output script will help give clues on how to solve your problem.

closed 2024-05-01 by evereux

ozserhatt, 2024-05-01:
Thank you for your answering, and your suggestion. I solved my problem. 

warm regards ..