evereux/pycatia #210: [FEATURE REQUEST] :

RoshanArute, 2024-05-18:
Hi,

I would like to use python to create CATAnalysis document pycatia. Currently, CATAnalysis are not even in the document type that are recognized. Hence we can not create a add the document. But we already have all the required interfaces to handle analysis document scenario except its creation. 

At my side I had added CATAnalysis type in allowed_document_types  of pycatia/scripts/document_types.py. It successfuly creates the Analysis document and all other interfaces works as expected for same document.

So could you please add "CATAnalysis" type in allowed list of types ?

Thank you for your time and expertise.

Minor bug -
Typo error -additional "_" at end of import in pycatia/analysis_interfaces/analysis_manager.py

`  def import_(self, i_document_to_import: Document) -> None:
`

evereux, 2024-05-18:
I was planning on adding all the doc types I can figure out this weekend.

I'll take a look at that bug report too.

evereux, 2024-05-18:
> def import_(self, i_document_to_import: Document) -> None:

This wasn't a  typo. Whilst I _think_ you can use `import` as a class method name it's generally a bad idea. My IDEs really complain if I try.

evereux, 2024-05-20:
You can now add this document type in 0.6.9.

closed 2024-05-20 by evereux