evereux/pycatia #212: Shaft

hosseintfr, 2024-05-27:
I want to shaft (revolve) an sketch in Catia using pycatia. I want an example. Can you please help me?
I can pad, rib, pocket,... in Catia by Pycatia, but I cannot Shaft. 
My problem is getting error with defining revolution axis for shaft.

evereux, 2024-05-27:
Can you please provide your code.

hosseintfr, 2024-05-27:
No. I just want an example.
My codes are in company system and so we cannot share our codes. Sharing codes is illegal in my company.

evereux, 2024-06-01:
You can't provide a stripped down version of your code, separate from your code base, to help me support you? OK.

I can't always give examples for this free and open source software in my spare time every time someone just asks for it with or without a please or thankyou. 🙄

Now, that said, try this with version 0.7.0 as there were a number of bug fixes, one of which was a dependency that `Shaft` uses.

I'm closing this. If you're still having a problem you can re-open this but you're going to need to provide a small script that demonstrates your issue so I can help you.

closed 2024-06-01 by evereux