evereux/pycatia #213: [QUESTION] Active Document

ozserhatt, 2024-05-29:
Hello friends !
That's important question that I have:
We have a tree from Engine Product 
for example;
Product1
      Product2
            Part1
            Part2
      Product3
When I put this product tree in a loop, and when I come to the part one, how can I make it the active document part so that I can perform process it here ?

evereux, 2024-05-29:
What will the trigger be for you to start working on it? That fact that it's a CATPart? Check the products name?

There are other methods too, see the examples regarding products.

ozserhatt, 2024-05-29:
I can add the parameterset with parameters active document part. But I have a tree with engine product and parts inside the product. So mt active document product now. 

My purpose : I want to add the parameterset all CATParts which inside the product. I can loop the all products and parts. But I can sign and active when come to part so, I can not add parameters

evereux, 2024-05-29:
Please provide a simple stripped down version of your code showing only what you have described.

ozserhatt, 2024-05-30:
def create_part_object(product):
    pass
    ### I CAN NOT CREATE AN OBJECT FROM PART, BECAUSE ACTIVE DOCUMENT IS PRODUCT
    ### I CAN NOT ACTIVATE THE PART DOCUMENT

 

def traverse_product_structure(product, depth=0):
    children = product.get_children()
    for child in children:
        if child.reference_product.type == "CATProduct":
            print(child.file_name)
            traverse_product_structure(child, depth + 1)
        elif child.reference_product.type == "CATPart":
            child_part = child.reference_product.parent
            print(child_part.name)
            create_part_object(child)

 

caa = catia()
app = caa.application
doc = app.documents
active_document = ProductDocument(caa.active_document.com_object)
product = Product(active_document.product.com_object)
traverse_product_structure(product)

ozserhatt, 2024-05-30:
[image]

evereux, 2024-05-30:
I've edited your code sample so that it reads better and I can copy and paste into my editor without having to format it. Please take a look by editing your post to see how I did that for future reference.

Also, please include all the imports required so I can run your script.

I wasn't sure how to do this as I don't think I've done this before. However, here is what I've come up with:

```python

from pycatia import catia
from pycatia.in_interfaces.document import Document
from pycatia.in_interfaces.documents import Documents
from pycatia.product_structure_interfaces.product_document import ProductDocument
from pycatia.product_structure_interfaces.product import Product
from pycatia.mec_mod_interfaces.part import Part
from pycatia.mec_mod_interfaces.part_document import PartDocument

def work_in_part(part):
    part = Part(part.com_object)
    parameters = part.parameters
    parameters.create_dimension('length', 'Length', 100)
    

def traverse_product_structure(product: Product, documents: Documents, depth=0):
    children = product.get_children()
    for child in children:
        if child.reference_product.type == "CATProduct":
            print(child.file_name)
            traverse_product_structure(child, depth + 1)
        elif child.reference_product.type == "CATPart":
            print(child.reference_product.file_name)
            part_document = PartDocument(documents.item(child.reference_product.file_name).com_object)
            part = part_document.part
            work_in_part(part)

caa = catia()
application = caa.application
documents = application.documents
active_document = ProductDocument(application.active_document.com_object)
product = active_document.product
traverse_product_structure(product, documents)

```

What I've basically done here is 
* pass the `Documents` collection to your `traverse_product_structure` function.
* if the `Product` is a CATPart, get it's filename and use that to get the `Part` object.

I don't know if this is how users typically do this. There is probably a better way. If I think if it I'll post back here.

Please note that if you have the same CATPart instantiated several times in the tree the `work_in_part` will duplicate that work.

Hope this makes sense!

ozserhatt, 2024-05-31:
It doesn't work :(
I shared a photo related to what I want to do in the previous comment, I hope you understood.
How can we do it ? :(

evereux, 2024-05-31:
I forgot to pass `documents` to the `traverse_product_structure` on line 20. This explains the error message you were getting.

I've edited my reply to fix this.

ozserhatt, 2024-05-31:
Good Job! That's OK, Thanks to a lot.. 😊

evereux, 2024-05-31:
Good stuff.

Don't forget, if a part is instantiated in several places you'll add duplicate parameters unless you check for the parameters existence first.

evereux, 2024-06-01:
Closing as issue seems to be resolved.

closed 2024-06-01 by evereux