evereux/pycatia #215: [QUESTION] REMOVE THE PARAMETERSET ???

ozserhatt, 2024-06-04:
from pycatia import catia
from pycatia.mec_mod_interfaces.part import Part
from pycatia.mec_mod_interfaces.part_document import PartDocument
from pycatia.knowledge_interfaces.parameters import Parameters

caa = catia()
application = caa.application
document = PartDocument(caa.active_document.com_object)
part = Part(document.part.com_object)
bodies = part.bodies

# gets part parameters
part_parameters = part.parameters

# gets RootParameterSet inside of parameters
root_parameter_set = part_parameters.root_parameter_set
# ParameterSet(name="Parameters")

# gets ParameterSets inside of RootParameterSet
parameter_sets = root_parameter_set.parameter_sets

# gets all parameter sets  inside of parameter_sets
sub_parameter_set = Parameters(parameter_sets.parameter_sets)
# [ParameterSet(name="Parameters_Pad"), ParameterSet(name="Parameters_Pocket")]

for parm_set in sub_parameter_set:
    print(parm_set.name)

### HOW CAN I REMOVE THE PARAMETERSET ???``

ozserhatt, 2024-06-04:
[image]

evereux, 2024-06-04:
I've edited your post to format your code again. Please, edit your reply to see how I did that.

The Parameters object has a method called [remove](https://github.com/evereux/pycatia/blob/9f48c90e11539d3e6ae387ca3244a10e18ef8160/pycatia/knowledge_interfaces/parameters.py#L489). You should be able to pass it the parameter name or index to remove it.

evereux, 2024-06-12:
This deletes the `ParameterSet` called `delete_me`.

I worked this out by recording a macro and reading the generated code to give me clues. I can't stress enough how valuable a tool this is.

Couldn't see any other way to do it as the `ParameterSets` collection object has no method for deleting.

```python

from pycatia import catia
from pycatia.mec_mod_interfaces.part import Part
from pycatia.mec_mod_interfaces.part_document import PartDocument
from pycatia.knowledge_interfaces.parameters import Parameters

caa = catia()
application = caa.application
part_document = PartDocument(caa.active_document.com_object)
selection = part_document.selection

part = part_document.part
bodies = part.bodies

part_parameters = part.parameters

root_parameter_set = part_parameters.root_parameter_set

parameter_sets = root_parameter_set.parameter_sets

delete_me = parameter_sets.item("delete_me")

selection.clear()
selection.add(delete_me)
selection.delete()

```

ozserhatt, 2024-06-13:
Good stuff. That is a cleverly. :) Well done!

closed 2024-06-14 by evereux